tesseract-robotics/tesseract_planning #16: Integrate TOTG

mpowelson, 2021-01-18:
Integrated into MoveIt [HERE](https://github.com/ros-planning/moveit/pull/1365/files). Looks like it is mostly Eigen types and should not be difficult to port.

mpowelson, 2021-01-22:
@Levi-Armstrong TOTG looks very promising. Here is a comparison of ISP results to TOTG results for freespace example. These are rough plots of the accelerations. I attached the csv files if you want to do better plotting. The columns are position, velocity, acceleration, time.
[freespace_plan_results_isp.csv.txt](https://github.com/ros-industrial-consortium/tesseract_planning/files/5857341/freespace_plan_results_isp.csv.txt)
[freespace_plan_results_totg.csv.txt](https://github.com/ros-industrial-consortium/tesseract_planning/files/5857342/freespace_plan_results_totg.csv.txt)

ISP
[image]

TOTG
[image]

Levi-Armstrong, 2021-01-22:
Looks good.

PR #23 (2021-01-23): Port TOTG
Port TOTG from MoveIt to Tesseract.

Closes #16
closed 2021-01-23 by Levi-Armstrong