tesseract-robotics/tesseract_planning #24: Port this MoveIt PR Constrained OMPL Planner

Levi-Armstrong, 2021-01-22:
This [PR](https://github.com/ros-planning/moveit/pull/2273) should be straight forward port since we recently had something similar in master.

chriswang2010, 2022-09-28:
hi @Levi-Armstrong , is this feature porting still in the plan? Thanks!

Levi-Armstrong, 2022-09-29:
The port was done a while back but is on an older version of Tesseract. It did not perform well for glass upright example so I have not revived it yet. I can dig up the branch if you are interested in updating it to the latest and testing it out.

Levi-Armstrong, 2022-09-29:
https://github.com/tesseract-robotics/tesseract_planning/blob/9873a4f88b85d4426f6a525ce207e0b66dcfafbd/tesseract_motion_planners/src/ompl/config/ompl_planner_constrained_config.cpp

https://github.com/tesseract-robotics/tesseract_planning/tree/9873a4f88b85d4426f6a525ce207e0b66dcfafbd/tesseract_motion_planners/src/ompl/config

chriswang2010, 2022-09-29:
Thanks! I will take a look at it and test it out!

Levi-Armstrong, 2022-09-30:
The main issue with this approach is that it samples in joint space then uses gradient decent to drive the state to the constraint manifold. This is what was making it take for ever. In the case of the glass upright problem it would probably be better to do sampling in cartesian space and solve IK which should be much faster.