tesseract-robotics/tesseract_planning #27: TOTG Error: Error while integrating backward: Negative path velocity

mpowelson, 2021-01-25:
I assume this is related to https://github.com/ros-planning/moveit/issues/2495. Wherever this gets addressed first, we need to port the fix or post it over there.

Levi-Armstrong, 2021-01-27:
This issues is most likely due to uninitialized data [here](https://github.com/ros-industrial-consortium/tesseract_planning/blob/2c61eba797a158eb5775f44adaf15e0d6426c9a2/tesseract_time_parameterization/src/time_optimal_trajectory_generation.cpp#L774) which can used. The question is what should it be initialized to?

Levi-Armstrong, 2021-01-27:
Also noticed that there is several `<=` and `>=` when comparing floats. Some places it leverages and EPS to account for numerical issues when comparing floats, but not all places.

mpowelson, 2021-01-27:
Odd the clang tidy build didn't fail. Yeah. I also seem to remember seeing somewhere where I thought there was a memory leak. This implementation probably needs a general cleanup

mpowelson, 2021-01-28:
There is an assert checking the condition where it would be unitialized. I'm actually getting this pretty frequently now. However, that assert is not triggered when it happens

mpowelson, 2021-01-28:
I'm not sure of the root cause, but it seems that a workaround is to append an extra joint that just increments such that no point ever repeats.

This passes
```
TEST(time_optimal_trajectory_generation, test_return_home)
{
  Eigen::VectorXd waypoint(7);
  std::list<Eigen::VectorXd> waypoints;

  waypoint << 0, 0.7, -2.1, 0, -0.25, 0,1;
  waypoints.push_back(waypoint);
  waypoint << 0, 0, 0, 0, 0, 0,2;
  waypoints.push_back(waypoint);
  waypoint << 0, 0.70001, -2.1, 0, -0.25, 0,3;
  waypoints.push_back(waypoint);
  waypoint << 0, 0, 0, 0, 0, 0.1, 4;
  waypoints.push_back(waypoint);

  Eigen::VectorXd max_velocities(7);
  max_velocities.setOnes();
  Eigen::VectorXd max_accelerations(7);
  max_accelerations.setOnes();

  Trajectory trajectory(Path(waypoints, 0.001), max_velocities, max_accelerations, 0.001);
  EXPECT_TRUE(trajectory.isValid());
}
```

mpowelson, 2021-02-02:
A workaround has been added in #32. However the underlying issue is still present. @Levi-Armstrong should we close this or leave it open? Unless I run into more problems, I don't intend to look into this further. While I'd definitely prefer we get to the bottom of it at some point, I'm thinking that effort might be better put towards integrating TOPPRA.