tesseract-robotics/tesseract_planning #33: New information not getting copied in instructions

mpowelson, 2021-01-29:
Something that I have run into multiple times now is that when new things get added to instructions, places where they get copied that information is lost. An example is something like this
```
      CompositeInstruction new_cc(cc->getProfile(), cc->getOrder(), cc->getManipulatorInfo());
      new_cc.setDescription(cc->getDescription());
      new_cc.setStartInstruction(cc->getStartInstruction());
  ```
If we add new_item to CompositeInstruction, we have to go through and add `new_cc.new_item = cc->new_item` in a lot of places - inevitably missing one and introducing a bug. I would think we should just be using a copy constructor in all of these cases, but perhaps there is something preventing this.

Levi-Armstrong, 2021-01-29:
I do not see a custom copy or assignment constructor in the CompositeInstruction or am I missing something?

mpowelson, 2021-01-29:
I don't think you are. So in this case we should just be doing `CompositeInstruction new_cc(*cc)`? We have code like this all over the place, but it might be a fossil.

mpowelson, 2021-03-23:
I think this was integrated into some of the past few PRs.

closed 2021-03-23 by mpowelson