tesseract-robotics/tesseract_planning #43: Profile Compile time

mpowelson, 2021-02-27:
It might be interesting to profile the Tesseract compile time was done for MoveIt here: https://github.com/ros-planning/moveit/issues/2081

mpowelson, 2021-07-16:
I ran the profiling on tesseract_planning and the dependencies in dependencies.rosinstall. Here is the output. [Here](https://github.com/mpowelson/development_notes/blob/master/Code%20Optimization%20and%20Debugging/profiling_compile_times.md) are my notes on how to generate this. @Levi-Armstrong might be interested.

```
**** Time summary:
Compilation (538 times):
  Parsing (frontend):         1540.9 s
  Codegen & opts (backend):    526.4 s

**** Files that took longest to parse (compiler frontend):
 16728 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/CMakeFiles/tesseract_motion_planners_ompl.dir/src/ompl/profile/ompl_default_plan_profile.cpp.o
 16276 ms: /home/matthew/code/tesseract/build//trajopt/CMakeFiles/trajopt.dir/src/problem_description.cpp.o
 15829 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/CMakeFiles/tesseract_motion_planners_simple.dir/src/simple/profile/simple_planner_utils.cpp.o
 15750 ms: /home/matthew/code/tesseract/build//trajopt/CMakeFiles/trajopt.dir/src/collision_terms.cpp.o
 15739 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/CMakeFiles/tesseract_motion_planners_descartes.dir/src/descartes/profile/descartes_default_plan_profile.cpp.o
 15468 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/CMakeFiles/tesseract_motion_planners_descartes.dir/src/descartes/descartes_robot_sampler.cpp.o
 13781 ms: /home/matthew/code/tesseract/build//tesseract_kinematics/CMakeFiles/tesseract_kinematics_kdl.dir/src/kdl/kdl_fwd_kin_chain.cpp.o
 13092 ms: /home/matthew/code/tesseract/build//tesseract_process_managers/CMakeFiles/tesseract_process_managers.dir/src/core/process_planning_server.cpp.o
 12934 ms: /home/matthew/code/tesseract/build//tesseract_kinematics/CMakeFiles/tesseract_kinematics_kdl.dir/src/kdl/kdl_fwd_kin_tree.cpp.o
 12784 ms: /home/matthew/code/tesseract/build//tesseract_kinematics/CMakeFiles/tesseract_kinematics_opw.dir/src/opw/opw_inv_kin.cpp.o

**** Files that took longest to codegen (compiler backend):
 14329 ms: /home/matthew/code/tesseract/build//trajopt/CMakeFiles/trajopt.dir/src/problem_description.cpp.o
 11265 ms: /home/matthew/code/tesseract/build//trajopt/CMakeFiles/trajopt.dir/src/kinematic_terms.cpp.o
  8203 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/examples/CMakeFiles/tesseract_motion_planners_raster_example.dir/raster_example.cpp.o
  7847 ms: /home/matthew/code/tesseract/build//tesseract_urdf/CMakeFiles/tesseract_urdf.dir/src/convex_mesh.cpp.o
  7788 ms: /home/matthew/code/tesseract/build//trajopt/CMakeFiles/trajopt.dir/src/collision_terms.cpp.o
  7069 ms: /home/matthew/code/tesseract/build//trajopt_sco/CMakeFiles/trajopt_sco.dir/src/modeling_utils.cpp.o
  6860 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/examples/CMakeFiles/tesseract_motion_planners_chain_example.dir/chain_example.cpp.o
  6846 ms: /home/matthew/code/tesseract/build//tesseract_environment/CMakeFiles/tesseract_environment_core.dir/src/core/environment.cpp.o
  6242 ms: /home/matthew/code/tesseract/build//tesseract_process_managers/CMakeFiles/tesseract_process_managers.dir/src/task_generators/fix_state_collision_task_generator.cpp.o
  6091 ms: /home/matthew/code/tesseract/build//tesseract_motion_planners/CMakeFiles/tesseract_motion_planners_ompl.dir/src/ompl/profile/ompl_default_plan_profile.cpp.o

**** Templates that took longest to instantiate:
 45536 ms: Eigen::JacobiSVD<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 2>::Jacob... (21 times, avg 2168 ms)
 45420 ms: Eigen::JacobiSVD<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 2>::compute (21 times, avg 2162 ms)
 41087 ms: Eigen::internal::qr_preconditioner_impl<Eigen::Matrix<double, -1, -1... (21 times, avg 1956 ms)
 38175 ms: Eigen::Transform<double, 3, 1, 0>::inverse (108 times, avg 353 ms)
 26949 ms: Eigen::EigenSolver<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::Eigen... (21 times, avg 1283 ms)
 26850 ms: Eigen::EigenSolver<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::compu... (21 times, avg 1278 ms)
 25876 ms: Eigen::Transform<double, 3, 1, 0>::operator* (131 times, avg 197 ms)
 25290 ms: Eigen::internal::transform_transform_product_impl<Eigen::Transform<d... (123 times, avg 205 ms)
 24900 ms: boost::breadth_first_search<boost::adjacency_list<boost::listS, boos... (133 times, avg 187 ms)
 24836 ms: boost::detail::bfs_dispatch<boost::associative_property_map<std::map... (133 times, avg 186 ms)
 24573 ms: boost::detail::bfs_helper<boost::adjacency_list<boost::listS, boost:... (133 times, avg 184 ms)
 24566 ms: Eigen::RealSchur<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::compute... (21 times, avg 1169 ms)
 23913 ms: Eigen::HouseholderSequence<Eigen::Matrix<double, -1, -1, 0, -1, -1>,... (21 times, avg 1138 ms)
 23874 ms: Eigen::internal::apply_block_householder_on_the_left<Eigen::Block<Ei... (21 times, avg 1136 ms)
 23418 ms: boost::breadth_first_search<boost::adjacency_list<boost::listS, boos... (133 times, avg 176 ms)
 23296 ms: boost::breadth_first_search<boost::adjacency_list<boost::listS, boos... (133 times, avg 175 ms)
 22156 ms: Eigen::Transform<double, 3, 1, 0>::rotation (47 times, avg 471 ms)
 22131 ms: Eigen::Transform<double, 3, 1, 0>::computeRotationScaling<Eigen::Mat... (47 times, avg 470 ms)
 22028 ms: Eigen::RealSchur<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::compute... (21 times, avg 1048 ms)
 21129 ms: boost::breadth_first_visit<boost::adjacency_list<boost::listS, boost... (133 times, avg 158 ms)
[... cut ...]
  box.cpp.o  (2024 ms)
  octomap.cpp.o  (1990 ms)
  mesh.cpp.o  (1929 ms)
  ...

61775 ms: /home/matthew/code/tesseract/devel/include/tesseract_common/types.h (included 176 times, avg 350 ms), included via:
  descartes_utils.cpp.o descartes_utils.h  (2077 ms)
  freespace_example.cpp.o  (1015 ms)
  memory_usage_example.cpp.o  (926 ms)
  kinematics_information.cpp.o kinematics_information.h  (714 ms)
  groups.cpp.o utils.h  (638 ms)
  raster_manager_example.cpp.o  (622 ms)
  ...

  done in 0.8s.

```

Levi-Armstrong, 2021-07-16:
Is there documentation or guidance on how you go about using this information to improve things?

mpowelson, 2021-07-16:
Not that I know of. I was mostly just running this to see if Taskflow showed up as a big bottleneck. I imagine the boost and eigen stuff we are kind of stuck with.

I found the expensive headers interesting. I imagine we could clean up some of those. I bet there are some places we can forward declare things to prevent including things like graph.h or environment.h

I also noticed that several of the files in tesseract_motion_planners showed up in longest to parse. I think we've known that we should probably split up that target regardless to avoid bottlenecks.