tesseract-robotics/tesseract_planning #56: How to set OMPL weights?

mpowelson, 2021-03-24:
At some point the OMPL planner got refactored to take a StateSamplerAllocator instead of a weights vector. Do we have an example of a StateSamplerAllocator that generates a WeightedRealVectorStateSampler? If so, it might be nice to put that in a utility somewhere where users can just assign it.

This does bring up the difficulty that it assumes the same dof will always be used for this profile, but I don't know of an easy way around that.

Levi-Armstrong, 2021-03-26:
In the ompl_default_plan_profile.cpp there is an example of using the a weighted state sampler, let me know if this is not what you are looking for.

```
if (state_sampler_allocator)
    {
      rss->setStateSamplerAllocator(
          [=](const ompl::base::StateSpace* space) { return state_sampler_allocator(space, prob); });
    }
    else
    {
      Eigen::VectorXd weights = Eigen::VectorXd::Ones(dof);
      rss->setStateSamplerAllocator(
          std::bind(&allocWeightedRealVectorStateSampler, std::placeholders::_1, weights, limits));
    }
```

mpowelson, 2021-03-29:
Ok. So I think this would work. Is this the expected usage?

```
OMPLDefaultPlanProfile profile;
profile.state_sampler_allocator = [&](const ompl::base::StateSpace* space, const OMPLProblem& problem) { 
  Eigen::VectorXd my_weights = ...
  return WeightedRealVectorStateSampler(space, my_weights, problem.manip_fwd_kin->getLimits().joint_limits);}
} 
```

Levi-Armstrong, 2021-03-29:
Yea, I believe so.

mpowelson, 2021-04-01:
Ok. I might make a utility that returns the correct function.