tesseract-robotics/tesseract_planning #69: catkin_make error about tesseract_process_managers

CaffeMrDe, 2021-04-09:
==> Processing plain cmake package: 'tesseract_process_managers'
==> Building with env: '/home/fshs/tesseract_ros_ws/devel_isolated/tesseract_motion_planners/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/fshs/tesseract_ros_ws/build_isolated/tesseract_process_managers/devel'
==> make -j2 -l2 in '/home/fshs/tesseract_ros_ws/build_isolated/tesseract_process_managers/devel'
Scanning dependencies of target tesseract_process_managers
[  5%] Building CXX object CMakeFiles/tesseract_process_managers.dir/src/core/process_planning_server.cpp.o
[  5%] Building CXX object CMakeFiles/tesseract_process_managers.dir/src/task_generators/fix_state_bounds_task_generator.cpp.o
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/string:41,
                 from /usr/include/console_bridge/console.h:40,
                 from /home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/src/core/process_planning_server.cpp:29:
/usr/include/c++/9/ext/new_allocator.h: In instantiation of ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = tesseract_planning::DebugObserver; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = tesseract_planning::DebugObserver]’:
/usr/include/c++/9/bits/alloc_traits.h:482:2:   required from ‘static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = tesseract_planning::DebugObserver; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = tesseract_planning::DebugObserver; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<tesseract_planning::DebugObserver>]’
/usr/include/c++/9/bits/shared_ptr_base.h:548:39:   required from ‘std::_Sp_counted_ptr_inplace<_Tp, _Alloc, _Lp>::_Sp_counted_ptr_inplace(_Alloc, _Args&& ...) [with _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = tesseract_planning::DebugObserver; _Alloc = std::allocator<tesseract_planning::DebugObserver>; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/9/bits/shared_ptr_base.h:679:16:   required from ‘std::__shared_count<_Lp>::__shared_count(_Tp*&, std::_Sp_alloc_shared_tag<_Alloc>, _Args&& ...) [with _Tp = tesseract_planning::DebugObserver; _Alloc = std::allocator<tesseract_planning::DebugObserver>; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/9/bits/shared_ptr_base.h:1344:71:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<tesseract_planning::DebugObserver>; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = tesseract_planning::DebugObserver; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/9/bits/shared_ptr.h:359:59:   required from ‘std::shared_ptr<_Tp>::shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<tesseract_planning::DebugObserver>; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = tesseract_planning::DebugObserver]’
/usr/include/c++/9/bits/shared_ptr.h:701:14:   required from ‘std::shared_ptr<_Tp> std::allocate_shared(const _Alloc&, _Args&& ...) [with _Tp = tesseract_planning::DebugObserver; _Alloc = std::allocator<tesseract_planning::DebugObserver>; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}]’
/usr/include/c++/9/bits/shared_ptr.h:717:39:   required from ‘std::shared_ptr<_Tp> std::make_shared(_Args&& ...) [with _Tp = tesseract_planning::DebugObserver; _Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}]’
/opt/ros/noetic/include/taskflow/core/executor.hpp:549:40:   required from ‘std::shared_ptr<_Tp> tf::Executor::make_observer(ArgsT&& ...) [with Observer = tesseract_planning::DebugObserver; ArgsT = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}]’
/home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/src/core/process_planning_server.cpp:53:47:   required from here
/usr/include/c++/9/ext/new_allocator.h:145:20: error: invalid new-expression of abstract class type ‘tesseract_planning::DebugObserver’
  145 |  noexcept(noexcept(::new((void *)__p)
      |                    ^~~~~~~~~~~~~~~~~~
  146 |        _Up(std::forward<_Args>(__args)...)))
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/src/core/process_planning_server.cpp:34:
/home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/include/tesseract_process_managers/core/debug_observer.h:40:7: note:   because the following virtual functions are pure within ‘tesseract_planning::DebugObserver’:
   40 | class DebugObserver : public tf::ObserverInterface
      |       ^~~~~~~~~~~~~
In file included from /opt/ros/noetic/include/taskflow/core/executor.hpp:3,
                 from /opt/ros/noetic/include/taskflow/taskflow.hpp:3,
                 from /home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/include/tesseract_process_managers/core/process_planning_server.h:33,
                 from /home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/src/core/process_planning_server.cpp:33:
/opt/ros/noetic/include/taskflow/core/observer.hpp:191:16: note: 	‘virtual void tf::ObserverInterface::on_entry(tf::WorkerView, tf::TaskView)’
  191 |   virtual void on_entry(WorkerView w, TaskView task_view) = 0;
      |                ^~~~~~~~
/opt/ros/noetic/include/taskflow/core/observer.hpp:198:16: note: 	‘virtual void tf::ObserverInterface::on_exit(tf::WorkerView, tf::TaskView)’
  198 |   virtual void on_exit(WorkerView w, TaskView task_view) = 0;
      |                ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/string:41,
                 from /usr/include/console_bridge/console.h:40,
                 from /home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/src/core/process_planning_server.cpp:29:
/usr/include/c++/9/ext/new_allocator.h: In instantiation of ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = tesseract_planning::DebugObserver; _Args = {const char (&)[24]}; _Tp = tesseract_planning::DebugObserver]’:
/usr/include/c++/9/bits/alloc_traits.h:482:2:   required from ‘static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = tesseract_planning::DebugObserver; _Args = {const char (&)[24]}; _Tp = tesseract_planning::DebugObserver; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<tesseract_planning::DebugObserver>]’
/usr/include/c++/9/bits/shared_ptr_base.h:548:39:   required from ‘std::_Sp_counted_ptr_inplace<_Tp, _Alloc, _Lp>::_Sp_counted_ptr_inplace(_Alloc, _Args&& ...) [with _Args = {const char (&)[24]}; _Tp = tesseract_planning::DebugObserver; _Alloc = std::allocator<tesseract_planning::DebugObserver>; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/9/bits/shared_ptr_base.h:679:16:   required from ‘std::__shared_count<_Lp>::__shared_count(_Tp*&, std::_Sp_alloc_shared_tag<_Alloc>, _Args&& ...) [with _Tp = tesseract_planning::DebugObserver; _Alloc = std::allocator<tesseract_planning::DebugObserver>; _Args = {const char (&)[24]}; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/9/bits/shared_ptr_base.h:1344:71:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<tesseract_planning::DebugObserver>; _Args = {const char (&)[24]}; _Tp = tesseract_planning::DebugObserver; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
[... cut ...]
/opt/ros/noetic/include/taskflow/core/executor.hpp:549:40:   required from ‘std::shared_ptr<_Tp> tf::Executor::make_observer(ArgsT&& ...) [with Observer = tesseract_planning::DebugObserver; ArgsT = {const char (&)[24]}]’
/home/fshs/tesseract_ros_ws/src/tesseract_planning-master/tesseract_process_managers/src/core/process_planning_server.cpp:63:68:   required from here
/usr/include/c++/9/ext/new_allocator.h:145:20: error: invalid new-expression of abstract class type ‘tesseract_planning::DebugObserver’

CaffeMrDe, 2021-04-09:
process_planning_server.cpp line 53

 /** @todo Need to figure out if these can associated with an individual run versus global */
  //executor_->make_observer<DebugObserver>("ProcessPlanningObserver");

mpowelson, 2021-04-09:
Looks like you may be trying to build it with Taskflow 3 instead of 2. I actually just opened a pull request to update to 3.0 if you want to try that. #68

CaffeMrDe, 2021-04-10:
Oh ，the problem is  solved. Thank you  very much. You are the best!

closed 2021-04-12 by mpowelson