tesseract-robotics/tesseract_planning #70: Error building the repo `undefined reference to symbol 'pthread_rwlock_unlock@@GLIBC_2.2.5'`

mohamedsayed18, 2021-04-10:
I cloned the repo and its dependencies in the src of my workspace, and build it using `catkin build`
```
Errors     << tesseract_motion_planners:make /media/mohamed/C03CCDB43CCDA62E/work/tess_motion_space/logs/tesseract_motion_planners/build.make.000.log
/usr/bin/ld: CMakeFiles/tesseract_motion_planners_freespace_example.dir/freespace_example.cpp.o: undefined reference to symbol 'pthread_rwlock_unlock@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/tesseract_motion_planners_freespace_example.dir/build.make:208: examples/tesseract_motion_planners_freespace_example] Error 1
make[1]: *** [CMakeFiles/Makefile2:415: examples/CMakeFiles/tesseract_motion_planners_freespace_example.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:130: all] Error 2
cd /media/mohamed/C03CCDB43CCDA62E/work/Mx3D/tess_motion_space/build/tesseract_motion_planners; catkin build --get-env tesseract_motion_planners | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
.............................................................................................................................................
Failed     << tesseract_motion_planners:make                  [ Exited with code 2 ]                                                         
Failed    <<< tesseract_motion_planners                       [ 5 minutes and 17.1 seconds ]                                                 
Abandoned <<< tesseract_process_managers                      [ Unrelated job failed ]                                                       
[build] Summary: 25 of 27 packages succeeded.                                                                                                
[build]   Ignored:   None.                                                                                                                   
[build]   Warnings:  4 packages succeeded with warnings.                                                                                     
[build]   Abandoned: 1 packages were abandoned.                                                                                              
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 9 minutes and 36.0 seconds total.                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

mohamedsayed18, 2021-04-10:
I checked this [answer](https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line) related to pthread, and I tried to build with cmake argument 
`catkin build --catkin-make-args pthread` but the same error

Levi-Armstrong, 2021-04-12:
In tesseract_motion_planners CMakeLists.txt for the ompl and descartes libraries add `${CMAKE_THREAD_LIBS_INIT}` to the target_link_libraries list and see if it resolves the issue.

mohamedsayed18, 2021-04-12:
I added it to lines 71 and 104 in tesseract_motion_planners/CMakeLists.txt
```
target_link_libraries(${CMAKE_THREAD_LIBS_INIT} ${PROJECT_NAME}_ompl PUBLIC ${PROJECT_NAME}_core ${OMPL_LIBRARIES} Boost::boost)
target_link_libraries(${CMAKE_THREAD_LIBS_INIT} ${PROJECT_NAME}_descartes PUBLIC ${PROJECT_NAME}_core descartes::descartes_light descartes::descartes_samplers Boost::boost)
```
But still have the same error

Levi-Armstrong, 2021-04-12:
Ah, Also add find_package(Threads REQUIRED).

mohamedsayed18, 2021-04-12:
It worked, thank you so much for your quick responses

closed 2021-04-12 by mohamedsayed18