tesseract-robotics/tesseract_planning #79: Error compiling tesseract_process_managers

mohamedsayed18, 2021-05-02:
Error while building the repo
```
In file included from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/src/core/debug_observer.cpp:33:
/media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/include/tesseract_process_managers/core/debug_observer.h:54:8: error: ‘void tesseract_planning::DebugObserver::on_entry(tf::WorkerView, tf::TaskView)’ marked ‘final’, but is not virtual
   void on_entry(tf::WorkerView w, tf::TaskView tv) final;
        ^~~~~~~~
/media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/include/tesseract_process_managers/core/debug_observer.h:56:8: error: ‘void tesseract_planning::DebugObserver::on_exit(tf::WorkerView, tf::TaskView)’ marked ‘final’, but is not virtual
   void on_exit(tf::WorkerView w, tf::TaskView tv) final;
        ^~~~~~~
/media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/src/core/debug_observer.cpp:44:45: error: ‘w’ has incomplete type
 void DebugObserver::on_entry(tf::WorkerView w, tf::TaskView tv)
                              ~~~~~~~~~~~~~~~^
In file included from /usr/include/taskflow/core/graph.hpp:4,
                 from /usr/include/taskflow/core/task.hpp:3,
                 from /usr/include/taskflow/core/observer.hpp:3,
                 from /usr/include/taskflow/core/executor.hpp:5,
                 from /usr/include/taskflow/taskflow.hpp:4,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/include/tesseract_process_managers/core/debug_observer.h:31,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/src/core/debug_observer.cpp:33:
/usr/include/taskflow/declarations.hpp:19:7: note: forward declaration of ‘class tf::WorkerView’
 class WorkerView;
       ^~~~~~~~~~
/media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/src/core/debug_observer.cpp:49:44: error: ‘w’ has incomplete type
 void DebugObserver::on_exit(tf::WorkerView w, tf::TaskView tv)
                             ~~~~~~~~~~~~~~~^
In file included from /usr/include/taskflow/core/graph.hpp:4,
                 from /usr/include/taskflow/core/task.hpp:3,
                 from /usr/include/taskflow/core/observer.hpp:3,
                 from /usr/include/taskflow/core/executor.hpp:5,
                 from /usr/include/taskflow/taskflow.hpp:4,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/include/tesseract_process_managers/core/debug_observer.h:31,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract_planning/tesseract_process_managers/src/core/debug_observer.cpp:33:
/usr/include/taskflow/declarations.hpp:19:7: note: forward declaration of ‘class tf::WorkerView’
 class WorkerView;
       ^~~~~~~~~~
make[2]: *** [CMakeFiles/tesseract_process_managers.dir/build.make:76: CMakeFiles/tesseract_process_managers.dir/src/core/debug_observer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:132: CMakeFiles/tesseract_process_managers.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
cd /media/mohamed/C03CCDB43CCDA62E/work/tessract20/build/tesseract_process_managers; catkin build --get-env tesseract_process_managers | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
.............................................................................................................................................
Failed     << tesseract_process_managers:make                 [ Exited with code 2 ]                                                         
Failed    <<< tesseract_process_managers                      [ 23.9 seconds ]                                                               
Abandoned <<< tesseract_rosutils                              [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_monitoring                            [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_planning_server                       [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_ros_examples                          [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_rviz                                  [ Unrelated job failed ]                                                       
[build] Summary: 41 of 47 packages succeeded.                                                                                                
[build]   Ignored:   None.                                                                                                                   
[build]   Warnings:  14 packages succeeded with warnings.                                                                                    
[build]   Abandoned: 5 packages were abandoned.                                                                                              
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 14 minutes and 55.5 seconds total.                                                                                          
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

mohamedsayed18, 2021-05-02:
I was missing taskflow, I install it as `sudo apt install taskflow`  and it worked

closed 2021-05-02 by mohamedsayed18

Levi-Armstrong, 2021-05-02:
Ah, we upgraded the version of taskflow to 3.0.0 which broke the API. As you found I pushed new version for 18.04 and 20.04 to the ppa which an upgrade should solve the issue. Another option for reference if you are using ROS then on noetic you can also install ros-noetic-taskflow for 20.04.