tesseract-robotics/tesseract_planning #83: Tesseract_planning documentation

mohamedsayed18, 2021-05-20:
Is there a documentaion(doxygen generated, tutrorials) for the tesseract_planning?
I found this [page](https://ros-industrial-consortium.github.io/tesseract/doxygen/html/index.html) but it doesn't contain tesseract_motion_planners

Levi-Armstrong, 2021-05-20:
Sorry not yet. The best place to start is the examples, they have all been updated to leverage the new planning framework.

https://github.com/ros-industrial-consortium/tesseract_planning/tree/master/tesseract_motion_planners/examples
https://github.com/ros-industrial-consortium/tesseract_planning/tree/master/tesseract_process_managers/examples
https://github.com/ros-industrial-consortium/tesseract_ros/tree/master/tesseract_ros_examples/src
https://github.com/Levi-Armstrong/tesseract_ros_workcell

Levi-Armstrong, 2021-05-20:
I try to keep the tesseract_ros_workcell up-to-date but let me know if you run into any issues.

mohamedsayed18, 2021-05-20:
Do the examples in [tesseract_motion_planners](https://github.com/ros-industrial-consortium/tesseract_planning/tree/master/tesseract_motion_planners/examples) and [tesseract_process_managers](https://github.com/ros-industrial-consortium/tesseract_planning/tree/master/tesseract_process_managers/examples) have a corresponding launch?
I tried to run their executable as `./tesseract_motion_planners_chain_example` output:
```
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
Warning: Failed to load symbol 'TesseractIgnitionVisualizationPlugin'
         at line 62 in /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_visualization/src/visualization_loader.cpp
```
The rest of the output is repeated as follows, there is no visualization:
```
Info:    RRTConnect: Created 15 states (6 start + 9 goal)
Debug:   ParallelPlan.solveOne: Solution found by RRTConnect in 0.636221 seconds
Info:    ParallelPlan::solve(): Solution found by one or more threads in 0.636391 seconds
```
I will check the tesseract_ros_workcell, thanks for sharing.
I was working with the tesseract_ros_examples but there are some stuff that I can't understand yet, Is there another platform I should use for posting my questions instead of the issues?

Levi-Armstrong, 2021-05-20:
> I was working with the tesseract_ros_examples but there are some stuff that I can't understand yet, Is there another platform I should use for posting my questions instead of the issues?

You can use the Discussions: https://github.com/ros-industrial-consortium/tesseract_planning/discussions