tesseract-robotics/tesseract_planning #86: OSQP compile error with Windows ninja build

johnwason, 2021-05-27:
I tracked down the OSQP build error on Windows. The OSQP library and related QDLDL library are renaming the static library target to have the same name as the shared target. This is fine on Linux, since shared and static libraries have different extensions. On Windows, static libraries have a `.lib` extension, while shared libraries have a `.dll` extension, but also produce a `.lib` file that contains linking information. (Linux and OSX don't have this second linker file. They just use the `.so` or `.dylib` file.) Renaming to have the same name between static and shared results in a name conflict. Before this error was being silently ignored, but it really should have not been allowed.

The offending lines are here:

https://github.com/oxfordcontrol/osqp/blob/0b34f2ef5c5eec314e7945762e1c8167e937afbd/CMakeLists.txt#L268
https://github.com/oxfordcontrol/qdldl/blob/7d16b70a10a152682204d745d814b6eb63dc5cd2/CMakeLists.txt#L107

The current build configuration doesn't export symbols properly, so the shared libraries are not usable. It is necessary to add the 
`-DCMAKE_WINDOWS_EXPORT_ALL_SYMBOLS:BOOL=TRUE` cmake arg to produce symbols.

I created patched forks. I may clean the patches up and submit pull requests.

https://github.com/oxfordcontrol/osqp
https://github.com/johnwason/qdldl/

The fixed CMakeLists.txt file to use these patches:

```
cmake_minimum_required(VERSION 3.5.0)
project(osqp)

find_package(osqp QUIET)

if (NOT ${osqp_FOUND})
  message(WARNING "No valid OSQP version found. Cloning OSQP 0.6.0 into build directory")

  include(ExternalProject)

  ExternalProject_Add(osqp
    GIT_REPOSITORY    https://github.com/johnwason/osqp
    GIT_TAG           v0.6.2dup_name_fix
    SOURCE_DIR        ${CMAKE_BINARY_DIR}/../trajopt_ext/osqp-src
    BINARY_DIR        ${CMAKE_BINARY_DIR}/../trajopt_ext/osqp-build
    CMAKE_CACHE_ARGS
            -DCMAKE_INSTALL_PREFIX:STRING=${CMAKE_INSTALL_PREFIX}
            -DCMAKE_BUILD_TYPE:STRING=Release
            -DCMAKE_WINDOWS_EXPORT_ALL_SYMBOLS:BOOL=TRUE
  )
endif()

install(FILES package.xml DESTINATION share/osqp)
```

Conda already has an OSQP package so hopefully I can just use it.

I am reporting the solution but don't plan to submit a pull request to fix this issue.

rjoomen, 2023-08-22:
Has this been solved?

johnwason, 2023-08-23:
This was fixed with a patch in the trajopt repo: https://github.com/tesseract-robotics/trajopt/pull/287

closed 2023-08-23 by johnwason