tesseract-robotics/tesseract_planning #96: catkin build error problem:

henryhongyan, 2021-06-24:


henryhongyan, 2021-06-24:
System :ubutu18.04.1

I am using the catkin build command to compile the tesseract_ros package. I have downloaded the dependencies in the dependencies.rosinstall file and added tesseract_ext. However, the following problems occurred during the compilation process:

CMake Error at /home/ldf/tesseract_ros_ws/src/tesseract_planning/tesseract_motion_planners/CMakeLists.txt:15 (find_package):
  By not providing "Finddescartes_samplers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "descartes_samplers", but CMake did not find one.

  Could not find a package configuration file provided by
  "descartes_samplers" with any of the following names:

    descartes_samplersConfig.cmake
    descartes_samplers-config.cmake

  Add the installation prefix of "descartes_samplers" to CMAKE_PREFIX_PATH or
  set "descartes_samplers_DIR" to a directory containing one of the above
  files.  If "descartes_samplers" provides a separate development package or
  SDK, be sure it has been installed.

I tried to search and download descartes_samplers online, but I could not find it.
Is there a way to solve this problem? Thank you

marip8, 2021-06-24:
The `descartes_light` repository is still under quite a bit of development, so its current master branch is now incompatible with this repository. You should check out [tag 2.1 of `descartes_light`](https://github.com/swri-robotics/descartes_light/tree/0.2.1) as specified in this repository's [`dependencies.rosinstall` file](https://github.com/ros-industrial-consortium/tesseract_planning/blob/5b09248852f861c57dbdb20e5c9cabe6647e957a/dependencies.rosinstall#L13-L16)

henryhongyan, 2021-06-24:
Thank you for your reply  @marip8,
According to the method you gave, I can already find "descartes_samplers". Thanks again.

closed 2021-06-24 by henryhongyan