tesseract-robotics/tesseract_planning #109: Compiler errors when cloning from dependencies.rosinstall

dignakov, 2021-08-07:
I'm build tesseract following the instructions here: https://tesseract-docs.readthedocs.io/en/latest/index.html

After cloning the required repos and installing dependencies, I was getting quite a few compiler errors until I switched all of the repos to master (now there's just one).

It seems like the rosinstall file is pointing to versions that don't build together.

Levi-Armstrong, 2021-08-07:
Most likely you have the wrong branches. I recommend checking the rosinstall file in the repository and check the branch for tesseract_planning and tesseract_ros. All masters branches may not be synced.

dignakov, 2021-08-07:
Thanks, I tried just now manually setting all the commits/tags to what's in `tesseract_planning/dependencies.rosinstall`:

```
- git:
    local-name: ros_industrial_cmake_boilerplate
    uri: https://github.com/ros-industrial/ros_industrial_cmake_boilerplate.git
    version: 0.2.11
- git:
    local-name: tesseract
    uri: https://github.com/ros-industrial-consortium/tesseract.git
    version: 687e8906d3d29ff361040dbc0d7ad5e9abb22f27
- git:
    local-name: trajopt
    uri: https://github.com/ros-industrial-consortium/trajopt_ros.git
    version: 9502162d5ca110de9a2dc460156c446ec0d6c626
- git:
    local-name: descartes_light
    uri: https://github.com/swri-robotics/descartes_light.git
    version: 6b28e2d5c6a10fb4e2fd6f50ff83f0f07cb01146
- git:
    local-name: opw_kinematics
    uri: https://github.com/Jmeyer1292/opw_kinematics.git
    version: 0.4.3
- git:
    local-name: ifopt
    uri: https://github.com/ethz-adrl/ifopt.git
    version: 27451ce3212d3c0ccd3a0db9a1ee9a9963466674
```

I'm getting a few instances of these errors when I build (clean build after removing `build` and `devel`):

```
error: ‘using element_type = const class tesseract_kinematics::InverseKinematics’ {aka ‘const class tesseract_kinematics::InverseKinematics’} has no member named ‘getRedundancyCapableJointIndices’
  414 |                                                                 prob.manip_inv_kin->getRedundancyCapableJointIndices());
```

```
 error: too many arguments to function ‘std::vector<Eigen::Matrix<Scalar, -1, 1> > tesseract_kinematics::getRedundantSolutions(const Eigen::Ref<const Eigen::Matrix<Scalar, -1, 1> >&, const MatrixX2d&) [with FloatType = float; typename Eigen::internal::conditional<const Eigen::Matrix<Scalar, -1, 1>::IsVectorAtCompileTime, Eigen::InnerStride<1>, Eigen::OuterStride<> >::type = Eigen::InnerStride<1>; Eigen::MatrixX2d = Eigen::Matrix<double, -1, 2>]’
  143 |           tesseract_kinematics::getRedundantSolutions<FloatType>(*(sample.state), limits, redundancy_capable_joints);
```

```
 error: ‘_OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation) [with _IIter = __gnu_cxx::__normal_iterator<const Eigen::Matrix<float, -1, 1>*, std::vector<Eigen::Matrix<float, -1, 1>, std::allocator<Eigen::Matrix<float, -1, 1> > > >; _OIter = std::back_insert_iterator<std::vector<descartes_light::StateSample<float>, std::allocator<descartes_light::StateSample<float> > > >; _UnaryOperation = tesseract_planning::DescartesRobotSampler<FloatType>::sample() const [with FloatType = float]::<lambda(const auto:31&)>]’, declared using local type ‘tesseract_planning::DescartesRobotSampler<FloatType>::sample() const [with FloatType = float]::<lambda(const auto:31&)>’, is used but never defined [-fpermissive]
```

I'm on Ubuntu 20.04, ROS Noetic.

Full output:
```

Errors     << tesseract_motion_planners:make /home/dmitri/catkin/tesseract_branches/logs/tesseract_motion_planners/build.make.001.log
/home/dmitri/catkin/tesseract_branches/src/tesseract_planning/tesseract_motion_planners/src/trajopt_ifopt/trajopt_ifopt_motion_planner.cpp: In member function ‘virtual tesseract_common::StatusCode tesseract_planning::TrajOptIfoptMotionPlanner::solve(const tesseract_planning::PlannerRequest&, tesseract_planning::PlannerResponse&, bool) const’:
/home/dmitri/catkin/tesseract_branches/src/tesseract_planning/tesseract_motion_planners/src/trajopt_ifopt/trajopt_ifopt_motion_planner.cpp:133:55: warning: ‘void OsqpEigen::Settings::setMaxIteraction(int)’ is deprecated: Use setMaxIteration(int) instead. [-Wdeprecated-declarations]
  133 |   qp_solver->solver_.settings()->setMaxIteraction(8192);
      |                                                       ^
In file included from /home/dmitri/catkin/tesseract_branches/devel/include/OsqpEigen/OsqpEigen.h:12,
                 from /home/dmitri/catkin/tesseract_branches/devel/include/trajopt_sqp/osqp_eigen_solver.h:31,
                 from /home/dmitri/catkin/tesseract_branches/src/tesseract_planning/tesseract_motion_planners/src/trajopt_ifopt/trajopt_ifopt_motion_planner.cpp:31:
/home/dmitri/catkin/tesseract_branches/devel/include/OsqpEigen/Settings.hpp:92:66: note: declared here
   92 |         [[deprecated("Use setMaxIteration(int) instead.")]] void setMaxIteraction(const int maxIteration);
      |                                                                  ^~~~~~~~~~~~~~~~
/home/dmitri/catkin/tesseract_branches/src/tesseract_planning/tesseract_motion_planners/src/ompl/profile/ompl_default_plan_profile.cpp: In member function ‘virtual void tesseract_planning::OMPLDefaultPlanProfile::applyGoalStates(tesseract_planning::OMPLProblem&, const Isometry3d&, const tesseract_planning::Instruction&, const ManipulatorInfo&, const std::vector<std::__cxx11::basic_string<char> >&, int) const’:
/home/dmitri/catkin/tesseract_branches/src/tesseract_planning/tesseract_motion_planners/src/ompl/profile/ompl_default_plan_profile.cpp:414:85: error: ‘using element_type = const class tesseract_kinematics::InverseKinematics’ {aka ‘const class tesseract_kinematics::InverseKinematics’} has no member named ‘getRedundancyCapableJointIndices’
  414 |                                                                 prob.manip_inv_kin->getRedundancyCapableJointIndices());
      |                                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:331: CMakeFiles/tesseract_motion_planners_simple.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/tesseract_motion_planners_descartes.dir/build.make:102: CMakeFiles/tesseract_motion_planners_descartes.dir/src/descartes/descartes_robot_sampler.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:250: CMakeFiles/tesseract_motion_planners_descartes.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:304: CMakeFiles/tesseract_motion_planners_ompl.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

Levi-Armstrong, 2021-08-07:
I just triggered a CI build to see if something changed in the dependencies.

dignakov, 2021-08-07:
After trying to figure out what I was doing wrong for a while longer, I uninstalled ros-noetic-* and reinstalled, from scratch and everything seems to compile successfully.

I still don't know what it was that I had on there that was causing this, but it seems to compile.

Thank you for the help!

closed 2021-08-07 by dignakov