tesseract-robotics/tesseract_planning #110: compiler selecting the wrong overload of getLinkChildrenNames

dignakov, 2021-08-07:
I'm currently running into this error:

```
Errors     << tesseract_environment:make /home/dmitri/catkin/tesseract/logs/tesseract_environment/build.make.004.log
/home/dmitri/catkin/tesseract/src/tesseract/tesseract_environment/src/core/environment.cpp: In member function ‘virtual std::vector<std::__cxx11::basic_string<char> > tesseract_environment::Environment::getActiveLinkNames(const std::vector<std::__cxx11::basic_string<char> >&) const’:
/home/dmitri/catkin/tesseract/src/tesseract/tesseract_environment/src/core/environment.cpp:437:76: error: cannot convert ‘const std::vector<std::__cxx11::basic_string<char> >’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
  437 |   std::vector<std::string> ret = scene_graph_const_->getJointChildrenNames(joint_names);
      |                                                                            ^~~~~~~~~~~
      |                                                                            |
      |                                                                            const std::vector<std::__cxx11::basic_string<char> >
In file included from /home/dmitri/catkin/tesseract/src/tesseract/tesseract_environment/include/tesseract_environment/core/types.h:39,
                 from /home/dmitri/catkin/tesseract/src/tesseract/tesseract_environment/include/tesseract_environment/core/environment.h:37,
                 from /home/dmitri/catkin/tesseract/src/tesseract/tesseract_environment/src/core/environment.cpp:27:
/opt/ros/noetic/include/tesseract_scene_graph/graph.h:429:69: note:   initializing argument 1 of ‘std::vector<std::__cxx11::basic_string<char> > tesseract_scene_graph::SceneGraph::getJointChildrenNames(const string&) const’
  429 |   std::vector<std::string> getJointChildrenNames(const std::string& name) const;
      |                                                  ~~~~~~~~~~~~~~~~~~~^~~~
make[2]: *** [CMakeFiles/tesseract_environment_core.dir/build.make:63: CMakeFiles/tesseract_environment_core.dir/src/core/environment.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:134: CMakeFiles/tesseract_environment_core.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

It seems that the compiler is selecting the wrong overload for `getLinkNames`

I'm not sure if I'm building tesseract wrong, or if something else is wrong, but I followed the procedure on the wiki except switching all the repos to master.

Levi-Armstrong, 2021-08-07:
I believe you are using the master branch from tesseract. Check the tesseract_planning dependencies.rosinstall file for the correct hash. 687e8906d3d29ff361040dbc0d7ad5e9abb22f27

Levi-Armstrong, 2021-08-07:
you may also need to do the same for trajopt

dignakov, 2021-08-07:
Thanks, I just commented back in that issue, I'll rebuild one more time with that hash.

closed 2021-08-07 by dignakov