tesseract-robotics/tesseract_planning #111: Serialization of TrajOptDefaultSolverProfile missing

bi3ri, 2021-09-08:
Hei,

I find the serialization/deserialization feature quite handy, because right now I have to recompile my program for every little parameter change which takes ruffly one minute on my machine. TrajOptDefaultPlanProfile and TrajOptDefaultCompositeProfile have this feature implemented [here](https://github.com/ros-industrial-consortium/tesseract_planning/blob/master/tesseract_motion_planners/src/trajopt/serialize.cpp#L45) and [here](https://github.com/ros-industrial-consortium/tesseract_planning/blob/master/tesseract_motion_planners/src/trajopt/deserialize.cpp).

Is there a reason TrajOptDefaultSolverProfile is missing? Otherwise, I would try to create a PR.

Thanks,
Johannes

Levi-Armstrong, 2021-09-09:
We will very soon be updating these to leverage boost serialization to support importing and exporting profiles.

bi3ri, 2021-09-09:
Okay, thanks for the update :)

closed 2021-09-09 by bi3ri