tesseract-robotics/tesseract_planning #115: Update stable and unstable CI builds

marip8, 2021-09-13:
Now that `tesseract` is distributed on the ROS build farm, we should update the "stable" and "unstable" CI builds. It seems like the "stable" CI build should use the currently available `tesseract` debian packages rather than building `tesseract` from source at some specified commit. At the moment, this repository requires some changes made to `tesseract` that have not yet made it into the distributed debians which means all of `tesseract` needs to be built from source too

Levi-Armstrong, 2021-09-14:
The current CI builds the current state of the repository master branch leveraging the rosinstall file which is not intended to be compatible with the debian release of Tesseract. The current rosinstall tracks a specific commit of tesseract which it has been tested/upgraded to and the unstable was added to point to the master branch of Tesseract to known when breaking changes have occurred in Tesseract. Are suggesting creating a stable CI build which build a tesseract_planning commit using the debian? When I did the release of Tesseract I should have created a Tag in this repository and tesseract_ros.

marip8, 2021-09-15:
It would be extremely convenient if an easily identifiable version of this repository built with the distributed version of `tesseract`. An easy first step might be to create tags in this repository that correspond to those in `tesseract`, such that `tesseract_planning` x.y.z corresponds to `tesseract` x.y.z. It seems like other projects with multiple repositories take this approach, such as `boost`. This would help clarify what versions of each repository build with each other.

> Are suggesting creating a stable CI build which build a tesseract_planning commit using the debian?

I think that would be helpful, but maybe its really not necessary on second thought. My initial impression when I went to rebuild this repository the other day was that it would build with the distributed version of `tesseract` and it currently does not. Having a CI build to highlight that status would be nice, but having the corresponding tags might be enough to indicate what version of this repository builds with the distributed version of `tesseract`.

If we did add a build of the master branch of this repository against the distributed version of `tesseract`, I think we should make its success optional for merging new PRs. I think the important thing is that this repository tracks new changes to `tesseract` closely and we tag accordingly.

Levi-Armstrong, 2023-06-29:
I believe this is being done now.

closed 2023-06-29 by Levi-Armstrong