tesseract-robotics/tesseract_planning #127: Error:   checkKinematics: Manipulator translation norm

mohamedsayed18, 2021-11-02:
I'm having the following error and I'm trying to think what could be the reason behind it?
```
Error:   checkKinematics: Manipulator translation norm is greater than tolerance 0.001000!
         at line 77 in /media/mohamed/C03CCDB43CCDA62E/work/Mx3D/mx_space/devel/include/tesseract_kinematics/core/validate.h
```
I'm having a robot arm and an external axis. I'm trying to execute two StateWaypoints (start point, and another one). I'm just changing the value of one of the robot arm's joints.

Levi-Armstrong, 2021-11-03:
Do you have an example code that reproduces the error that I can troubleshoot?

mohamedsayed18, 2021-11-03:
> Do you have an example code that reproduces the error that I can troubleshoot?

https://github.com/mohamedsayed18/motion_planner/tree/positioner
The work is on the `abb_example` package
To launch the example `roslaunch abb_example m1_workcell.launch`

Levi-Armstrong, 2021-11-05:
I tested the kinematics in tesserat_ros_workcell for the robot with an external positioner and the validation passes. Are you using the the latest commits?

mohamedsayed18, 2021-11-05:
for the tesseract_ros_workcell it was working, it didn't show any errors but it took a long time to compute and I never tried to let the example run completely.
But in my example it still shows the validation error, I'll try to use the exact setup of the tesseract_ros_workcell problem and see what happens.
Note: Now, there are errors when building twc_motion_planning

Levi-Armstrong, 2021-11-06:
Ok, I will pull you workspace and look into it?

Levi-Armstrong, 2021-11-08:
I was able to recreate the issue this weekend and I am going to start tracking down the issue today.

Levi-Armstrong, 2021-11-09:
@mohamedsayed18 I opened a PR against your repository updating it to the latest. Also found a bug in opw_kinematics which I opened a PR on that repo and tesseract kinematic group which is resolved by [PR](https://github.com/tesseract-robotics/tesseract/pull/661)