tesseract-robotics/tesseract_planning #148: Running checkKinematics excessively

marrts, 2021-12-10:
Currently `checkKinematics` is run before every trajopt, descartes, and ompl task. With the latest changes to this function it can take a long time to run if you have a large number of tip links, working frames, and joints in your system since every possible combination is calculated. Perhaps it makes sense to separate the `checkKinematics` to run before all the other tasks. @marip8 had the idea to separate it into its own task as something that is optional to include. If we're confident no changes have been made then its a lot of excessive computation to be performing for a given motion plan. Perhaps the best solution involves modifying the `checkKinemtics` method to not have to calculate all these different combinations.

Levi-Armstrong, 2021-12-12:
I think that is resonable. This was mostly here during development to help catch breaking changes. I think it would be good to move it to its own task  and leave it up to the user to include if they want to.

marip8, 2021-12-13:
> Perhaps the best solution involves modifying the checkKinematics method to not have to calculate all these different combinations.

To clarify, all of the working frames and tool frames that are currently checked are attached by static transforms to the kinematic base frame and kinematic tip frame, respectively (to my knowledge). If that's the case, checking FK/IK for all of the combinations of both is redundant; it essentially is just checking that transformation math still works. I think we could modify `checkKinematics` to only check FK/IK with a single working frame and tool frame. If the check succeeds, that should be enough to say that the kinematics are defined correctly

Levi-Armstrong, 2022-01-05:
This has been addressed by #682

closed 2022-01-05 by Levi-Armstrong