tesseract-robotics/tesseract_planning #154: Add transition profile to PlanInstruction

Levi-Armstrong, 2022-01-05:
Currently all intermediate states generated for a given plan instruction get assigned the plan instructions profile. This is restrictive because you may only want that point to have the costs/constraints defined by the profile but all transitions points to have a different profile. Fore example, if you have a goal point which must be on a line but the path take to this goal point should be freespace there is currently no way to accomplish this with the current PlanInstruction using the process planner.  

I am going to update the PlanInstruction to have a getTransitionProfile and by default it will be the PlanInstruction profile.

@mpowelson @marip8 Let me know what you think.

marip8, 2022-01-07:
From our discussion this afternoon, it seems like we could solve this issue by changing the motion planners such that they produce both a motion plan as well as a revised input instruction for use in downstream planning stages. There are several use-cases where this would be beneficial, including:
1. `trajopt` MoveL interpolation (described in original post comment)
    - In the current `trajopt` planner, states that are interpolated between two input states get the same constraints as their parent waypoints. This can be a problem if the start and end waypoints need to be constrained tightly, but the interpolated points do not. This could be solved if some other planner (i.e. the simple planner) could perform interpolation and generate a new trajectory for `trajopt` in which the interpolated waypoints get assigned a different profile.
1. `descartes` with discrete Cartesian pose sampling followed by `trajopt` (@marrts)
    - If a `descartes` planner is allowed to sample Cartesian orientation by +/- 180 degrees about the z-axis, it could find a motion plan where the optimal solution is 180 degrees rotated from the input Cartesian waypoint. If this trajectory is passed to `trajopt` as a seed but the input instruction (i.e. tool path) remains the same, `trajopt` cannot solve appropriately because it has a fixed cost/constraint for the original, unrotated Cartesian pose. The easiest solution would be to allow the `descartes` planner to produce a new instruction where the appropriate poses are rotated. Another solution would be to create a constraint in `trajopt` that understands discrete pose sampling, but that seems fairly complicated in comparison.

Levi-Armstrong, 2022-01-19:
Implemented in #159

closed 2022-01-19 by Levi-Armstrong