tesseract-robotics/tesseract_planning #170: program terminated when using trajoptifopt

xibeisiber, 2022-02-10:
Hi, I pulled the latest tesseract-related packages and rebuild test_example mentioned in thie [issue](https://github.com/tesseract-robotics/tesseract_planning/issues/147).

If I use the trajoptIFopt solver, when the robot base is close to the collision cylinder (<0.75), the following error occurs:
`terminate called after throwing an instance of 'std::runtime_error'
  what():  Max error with buffer at T0 does not exist.
[test_example_test_example_node-3] process has died [pid 8363, exit code -6, cmd /home/jia/trajplan_ws/devel/lib/test_example/test_example_test_example_node __name:=test_example_test_example_node __log:=/home/jia/.ros/log/db513938-8a34-11ec-8da6-b07b25165f6d/test_example_test_example_node-3.log].`

If the distance between the robot base and cylinder is bigger than 0.8, it can find a valid path.

there's no such problem when using trajopt solver

tesseract_planning: db842b7f8075065c37d2b65404817e31f66219bf
tesseract_ros: e3375a54e91fc98a3da970dfe3537fb03bb9d7f4
tesseract: 31d3e1d6efbbd90decb09903605059055d27a061
trajopt: 1ccd4b3dfee656462e0fd4fa11f2dbfc109d8e93

Levi-Armstrong, 2022-03-27:
Thanks for reporting the issue.

Levi-Armstrong, 2022-03-28:
I was able to recreate the issue, but going to take some time to track down the issue.

closed 2025-01-04 by Levi-Armstrong