tesseract-robotics/tesseract_planning #177: Windows CI build failing

Levi-Armstrong, 2022-03-27:
@johnwason Have you seen this error before and have any idea what my be wrong?

``` bash
2022-03-27T06:02:05.2985230Z [  4%] Building CXX object CMakeFiles/tesseract_command_language.dir/src/core/instruction.cpp.obj
2022-03-27T06:02:05.3174592Z instruction.cpp
2022-03-27T06:02:06.4504376Z D:\a\tesseract_planning\tesseract_planning\src\tesseract_planning\tesseract_command_language\src\core\instruction.cpp(59): fatal error C1004: unexpected end-of-file found
2022-03-27T06:02:06.4697040Z NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~1\2019\ENTERP~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe' : return code '0x2'
2022-03-27T06:02:06.4698592Z Stop.
```

johnwason, 2022-03-27:
I think you need to update `dependencies_windows_build.rosinstall` to newer tags. I just built locally without error, so I suspect the version of tesseract you are cloning is too old.

Levi-Armstrong, 2022-03-27:
Ah, that makes sense and I forgot to check that.

Levi-Armstrong, 2022-03-28:
That was the issue.

closed 2022-03-28 by Levi-Armstrong