tesseract-robotics/tesseract_planning #191: TrajOptTaskflow Failure with new robot

cpetersmeier, 2022-05-02:
Hello everyone, 

unfortunately I am getting the same error message as in https://github.com/tesseract-robotics/tesseract_planning/issues/101.
```
Error:   TrajOptTaskflow Failure: 
         at line 51 in /path/to/workspace/src/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
```

However since my setup is a little different I am opening a new issue. 

I adapted the [basic_cartesian_example](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_examples/src/basic_cartesian_example.cpp) by changing the [robot configuration](https://github.com/tesseract-robotics/tesseract_ros/blob/9965197cdb2d5f775e26448e0fd2a9ee632a24e5/tesseract_ros_examples/launch/basic_cartesian_example.launch#L14) (I use a fanuc 6-DOF manipulator). For testing purposes I adapted the joint names, as well as the joint value data structure size;  I am only planning a point-to-point trajectory.

```
  Waypoint wp0 = StateWaypoint(joint_names, joint_pos);
  PlanInstruction start_instruction(wp0, PlanInstructionType::START);
  program.setStartInstruction(start_instruction);

  joint_pos(0) = 0.2;
  joint_pos(1) = 0.2;
  joint_pos(2) = 0.2;
  joint_pos(3) = 0.2;
  joint_pos(4) = 0.2;
  joint_pos(5) = 0.2;

  Waypoint wp1 = StateWaypoint(joint_names, joint_pos);

  // Plan freespace from start to point 1
  PlanInstruction plan_f_tmp(wp1, PlanInstructionType::FREESPACE, "freespace_profile");

  // Add Instructions to program
  program.push_back(plan_f_tmp);
```

I observed that the code runs without errors when I use e.g. the fanuc m20ia robot from [this](https://github.com/cpetersmeier/fanuc/commit/03944a6f3f8d4772337e3d6986dc40c7773d3518) repository.

However when I change to my own robot configuration (I used convex hull meshes of cad models with normals pointing outwards), the planning seems to almost complete successfully, but throws an error at the end (or sometimes in different scenarios earlier and therefore crashes).

[image]

Is there any way to mitigate this error message? I would also be very interested to know how to get more information about the reason the message comes up in the first place.

Thanks in advance!

### Commits:
`tesseract` [2fc91638e84197982430d0d0badcd88142f727e1](https://github.com/tesseract-robotics/tesseract/commit/2fc91638e84197982430d0d0badcd88142f727e1)
`tesseract_planning` [51817483fb813fa5eb357e6523b27201d2bb2da3](https://github.com/tesseract-robotics/tesseract_planning/commit/0b3c3fb44c627d8bd8689f0a13075de83c05269f)
`tesseract_ros` [6b9858b69105e9842cb556983050aca60f73d866](https://github.com/tesseract-robotics/tesseract_ros/commit/bcffbc6cfc7db0690cb69d5268b9724b9e35430a)

cpetersmeier, 2022-05-02:
Maybe related to this issue: I tried setting the `<arg name="ifopt" default="false"/>` to `true`. While it worked with the m20ia robot config, I got the following error using the new configuration:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Max error with buffer at T1 does not exist.
[tesseract_ros_examples_basic_cartesian_example_node-3] process has died [pid 55961, exit code -6, cmd /path/to/workspace/devel/lib/tesseract_ros_examples/tesseract_ros_examples_basic_cartesian_example_node __name:=tesseract_ros_examples_basic_cartesian_example_node __log:=/path/to/home/.ros/log/64d44114-ca13-11ec-befd-6b31c78c58c8/tesseract_ros_examples_basic_cartesian_example_node-3.log].
log file: /path/to/home/.ros/log/64d44114-ca13-11ec-befd-6b31c78c58c8/tesseract_ros_examples_basic_cartesian_example_node-3*.log
```

Levi-Armstrong, 2022-05-02:
I took a look at the srdf and this line is not correct. You will need to create your own kinematics plugin file for the robot and recommend using the opw inverse kinematics solver.

``` xml
<!--Groups kinematic plugins -->
    <kinematics_plugin_config filename="package://tesseract_support/urdf/lbr_iiwa_14_r820_plugins.yaml"/>
```

Levi-Armstrong, 2022-05-02:
> Maybe related to this issue: I tried setting the `<arg name="ifopt" default="false"/>` to `true`. While it worked with the m20ia robot config, I got the following error using the new configuration: [...]

This is currently an known issue and recommend not using the ifopt solver at this time.

cpetersmeier, 2022-05-03:
> I took a look at the srdf and this line is not correct. You will need to create your own kinematics plugin file for the robot and recommend using the opw inverse kinematics solver.
> 
> ```
> 
>     <kinematics_plugin_config filename="package://tesseract_support/urdf/lbr_iiwa_14_r820_plugins.yaml"/>
> ```

Thanks so much for the quick reply! I created a plugin file ...

`<kinematics_plugin_config filename="package://fanuc_m20id_support/urdf/m20id35_plugins.yaml"/>`

... and verified the OPW measurements against the mechanical robot documentation. The fwd vs. inv kinematics checks run correctly:

```
kinematic_plugins:
  search_paths:
    - /usr/local/lib
  search_libraries:
    - tesseract_kinematics_kdl_factories
  fwd_kin_plugins:
    manipulator:
      default: KDLFwdKinChain
      plugins:
        KDLFwdKinChain:
          class: KDLFwdKinChainFactory
          config:
            base_link: base_link
            tip_link: tool0
  inv_kin_plugins:
    manipulator:
      default: OPWInvKin
      plugins:
        OPWInvKin:
          class: OPWInvKinFactory
          config:
            base_link: base_link
            tip_link: tool0
            params:
              a1: 0.075   
              a2: -0.215 
              b: 0.0
              c1: 0.425
              c2: 0.840
              c3: 0.890
              c4: 0.09
              offsets: [0, 0, -1.5708, 0, 0, 0]
              sign_corrections: [1, 1, 1, 1, 1, 1]
```
Unfortunately, I am getting the same error message.

cpetersmeier, 2022-05-03:
Fortunately, I think the reason for the error originated out of the geometry model:

It was caused by `link_6` and `link_4` being in-collision at the start state of the trajectory. I used the `tesseract-ignition.tesseract-setup-wizard` to generate the collision matrix.

Since I used convex_hull approximations of the real geometries in my stl files, the two links are collision-free in the zero configuration (which is maybe used to generate the collision matrix?), but come in-collision once rotated around `joint_5`.

Levi-Armstrong, 2022-05-03:
That makes sense. The allowed collision matrix is generated by sampling the joint space at some resolution and if there are case where it is not in collision then it will not add it to the allowed collision matrix. It only adds things to the allowed collision matrix that are always in collision. If it is the case that link_4 and link_6 can never collide then it makes sense to disable it, if not you are able to set a different contact distance for link pairs.

cpetersmeier, 2022-05-04:
That sounds good, thank you very much for the quick reply!

closed 2022-05-04 by cpetersmeier