tesseract-robotics/tesseract_planning #195: How to configure semi-constrained Cartesian tasks

rhaschke, 2022-05-09:
Fig. 1 [here](https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications) suggests that it is possible to define semi-constrained Cartesian tasks. However, I couldn't find the corresponding API yet. Do you have a code example for this available?
Is it possible to define inequality constraints as well, e.g. constraining the end-effector position to a box (instead of a point)?

marip8, 2022-05-09:
Transferring issue to `tesseract_planning` as it's more relevant to the packages in this repo.

> Is it possible to define inequality constraints as well, e.g. constraining the end-effector position to a box (instead of a point)?

The `CartesianWaypoint` and `JointWaypoint` classes have [tolerances](https://github.com/tesseract-robotics/tesseract_planning/blob/196b39b77e9542aa5e8991e7426ba3972412576a/tesseract_command_language/include/tesseract_command_language/cartesian_waypoint.h#L165-L170) that you can specify for each dimension of that particular state space to get that behavior, but I don't think any of the planners acknowledge these tolerances at the moment. I think TrajOpt might support joint waypoint tolerances, but I'm fairly certain that it doesn't support Cartesian tolerances yet.

TrajOpt does expose the ability to weight particular state space dimensions. If any of these weights are set to zero, the constraint for that dimension is effectively ignored. I've used this a lot to get tool z-axis orientation freedom. I believe these weights are exposed [here in the TrajOpt planner plan profile](https://github.com/tesseract-robotics/tesseract_planning/blob/196b39b77e9542aa5e8991e7426ba3972412576a/tesseract_motion_planners/trajopt/include/tesseract_motion_planners/trajopt/profile/trajopt_default_plan_profile.h#L60-L61). Unfortunately this is a binary on/off solution but works well for some applications

Levi-Armstrong, 2022-05-10:
To add to @marip8 response. I believe there is a tolerance cartesian waypoint but it has had limited testing. We are in the process of migrating to TrajOptIFOPT which makes tolerance cost and constraints much easier, but it is still in testing.

rhaschke, 2022-05-29:
Thanks for your replies.

closed 2022-05-29 by rhaschke