tesseract-robotics/tesseract_planning #199: Adding a seed trajectory to a planning request

cpetersmeier, 2022-05-31:
Hello everyone,

I would like to generate a seed trajectory for trajopt using the result of a sampling based planner. I came across a couple of approaches and I am not sure which path to take.

- TrajOpt supports trajectory interpolation upon problem construction [here](https://github.com/tesseract-robotics/trajopt/blob/f79981f17feda7bdc12228536692c2af1008eaff/trajopt/src/problem_description.cpp#L347).

- There is a `SimplePlanner`, which has the corresponding planning profiles to interpolate between states in a composite instruction. This is used in the [chain example](https://github.com/tesseract-robotics/tesseract_planning/blob/57eaca720aca12423c61c14bf1251133bb6dc00e/tesseract_motion_planners/examples/chain_example.cpp#L143) to generate a seed trajectory ([deprecated](https://github.com/tesseract-robotics/tesseract_planning/blob/57eaca720aca12423c61c14bf1251133bb6dc00e/tesseract_motion_planners/core/include/tesseract_motion_planners/interface_utils.h#L42)?). However this example does not use the process planning server. Is there a recommended way to generate a seed using the process planning server as demonstrated in the `cartesian` example?

- The trajopt motion planner interface [bypasses](https://github.com/tesseract-robotics/tesseract_planning/blob/57eaca720aca12423c61c14bf1251133bb6dc00e/tesseract_motion_planners/trajopt/src/trajopt_motion_planner.cpp#L621) the trajopt-internal interpolation to generate an initial trajectory and uses a seed specified in the [request](https://github.com/tesseract-robotics/tesseract_planning/blob/57eaca720aca12423c61c14bf1251133bb6dc00e/tesseract_motion_planners/trajopt/src/trajopt_motion_planner.cpp#L289) object. I am curious, where the interpolation is taking place and how I would modify the seed to be a pre-calculated trajectory.

- Is the seed trajectory (and therefore also result trajectories) always of type composite instruction?

Thanks so much in advance!

Levi-Armstrong, 2022-05-31:
> There is a SimplePlanner, which has the corresponding planning profiles to interpolate between states in a composite instruction. This is used in the [chain [...]

The recommended approach is the process planning server. The simple planner is the seed generation which you can replace the simple planner with your approach to generating the seed.

> - The trajopt motion planner interface [...]

This is now accomplished by the simple planner to provide more control over how the seed is generated.

> Is the seed trajectory (and therefore also result trajectories) always of type composite instruction?

Yes, It should be a composite where each plan instruction in the input composite is a composite of move instructions.

In addition the simple planner will be skipped if you provided a filled out composite instruction in the seed of the input data structure.

Levi-Armstrong, 2022-05-31:
There are utility function for generating the seed data structure with empty composites along with creating two flat list between the input and the seed to make it easy to iterate over and populate.

cpetersmeier, 2022-06-02:
Thanks so much for the quick reply! That helped me out a lot. 

I would like to use a custom profile for the `SimplePlanner` in the `createTrajOptGenerator` Taskflow.
Assuming the `PlanInstructions` use a `"plan_profile"`, I added a simple planner profile like so:

```
planning_server.getProfiles()->addProfile<tesseract_planning::SimplePlannerLVSNoIKPlanProfile>(
    tesseract_planning::profile_ns::SIMPLE_DEFAULT_NAMESPACE, "plan_profile", simplePlannerProfile
);
```

However I get the following debug message:
```
Debug:   Profile 'plan_profile' was not found in namespace 'SIMPLE_PLANNER' for type 'N18tesseract_planning24SimplePlannerPlanProfileE'. Using default if available. Available profiles:
```

Levi-Armstrong, 2022-06-03:
This is a common issue when adding profiles. Make sure to use the base class as the template argument when adding the profile.  There is an open PR @marip8 which simplifies the profiles but currently only done for ompl. I hope we can get this new feature fully integrated in the near future but I know both myself and @marip8 have been busy.

cpetersmeier, 2022-06-07:
Awesome, thanks again for your reply!

May I ask how the seeds are handled with OMPL planners? Say the difference of 

- using a simple planner with two states only to convert the start and goal `PlanInstructions` to `MoveInstructions` as required by the OMPL planner.

- supplying a seed trajectory of e.g. 10 points between two states in a composite instruction. Do these correspond to graph nodes in e.g. a PRM planner, or would OMPL be invoked for each state of the seed trajectory?

Levi-Armstrong, 2022-06-07:
After a solution is found it leverages ompl utilities to try and resample the trajectory. It will always be equal or more than the original seed.

cpetersmeier, 2022-06-09:
Awesome, thanks a lot for the replies! Closing this.

closed 2022-06-09 by cpetersmeier