tesseract-robotics/tesseract_planning #206: Missing tesseract_command_language/core/serialization.h file when building Python interface

fishbotics, 2022-06-28:
Hi all, 

When I try to build the Python interface, I get the following error: 
```
tesseract_ws/src/tesseract_python/tesseract_python/swig/tesseract_command_language_python.i:98: Error: Unable to find 'tesseract_command_language/core/serialization.h'
```

It looks like `serialization.cpp` still exists in the `src` directory (https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_command_language/src/core/serialization.cpp) but is missing from the include directory as of  https://github.com/tesseract-robotics/tesseract_planning/commit/59cb07667c44fd22d2a81e7f7966a475b0d3c465

Should this file be removed from the Python interface as well? https://github.com/tesseract-robotics/tesseract_python/blob/master/tesseract_python/swig/tesseract_command_language_python.i#L41

Thanks!

PS Sorry if I'm posting these issues in the wrong repos. It's hard to decide which is the right place.

fishbotics, 2022-06-28:
I haven't used SWIG before, so I'm taking stabs in the dark here but I tried deleting lines 41 and 98 in that file and rebuilding and  I get a ton of errors that are maybe related and maybe not.

For example, 
```/home/fishy/tesseract_ws/devel/include/tesseract_common/types.h:248: Error: Syntax error in input(1).                                                                                                  
make[2]: *** [CMakeFiles/tesseract_command_language_python_swig_compilation.dir/build.make:64: CMakeFiles/_tesseract_command_language_python.dir/tesseract_command_language_pythonPYTHON.stamp] Error 1
make[2]: *** Deleting file 'CMakeFiles/_tesseract_command_language_python.dir/tesseract_command_language_pythonPYTHON.stamp'                                                                           
make[1]: *** [CMakeFiles/Makefile2:1046: CMakeFiles/tesseract_command_language_python_swig_compilation.dir/all] Error 2                                                                                
```

I didn't make any changes to that `types.h` file (it's not even in the `src` directory), but the line that it's complaining about is (marked below with `>>>>`):

```#include <boost/serialization/export.hpp>
#include <boost/serialization/tracking.hpp>
>>>> BOOST_CLASS_EXPORT_KEY2(tesseract_common::PluginInfo, "PluginInfo")
BOOST_CLASS_EXPORT_KEY2(tesseract_common::PluginInfoContainer, "PluginInfoContainer")
BOOST_CLASS_EXPORT_KEY2(tesseract_common::KinematicsPluginInfo, "KinematicsPluginInfo")
BOOST_CLASS_EXPORT_KEY2(tesseract_common::ContactManagersPluginInfo, "ContactManagersPluginInfo")
BOOST_CLASS_EXPORT_KEY2(tesseract_common::CalibrationInfo, "CalibrationInfo")

Levi-Armstrong, 2022-06-29:
Oh, you are trying to build the python bindings. I would jump over the tesseract_python repo and ask questions there. @johnwason is the author and maintainer.

fishbotics, 2022-06-29:
Ah, great. I'll move this issue over there. Thanks for all your help @Levi-Armstrong.

closed 2022-06-29 by fishbotics

johnwason, 2022-06-29:
We should add python pip instructions to the readme. Instructions are in this discussion: https://github.com/tesseract-robotics/tesseract/discussions/754