tesseract-robotics/tesseract_planning #211: Windows Compile Error

johnwason, 2022-06-30:
I am getting this compile error, and I am not sure what is causing it:

```
FAILED: trajopt_ifopt/CMakeFiles/tesseract_motion_planners_trajopt_ifopt.dir/src/trajopt_ifopt_motion_planner.cpp.obj 
C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBT_USE_DOUBLE_PRECISION -DBoost_USE_MULTITHREADED=ON -DBoost_USE_STATIC_LIBS=OFF -DBoost_USE_STATIC_RUNTIME=OFF -DDISABLE_PNG -DTESSERACT_ASSIMP_USE_PBRMATERIAL=1 -DTESSERACT_PARSE_POINT_CLOUDS=\"ON\" -DTINYXML2_IMPORT -D_USE_MATH_DEFINES=ON -D__SSE2__ -D__SSE3__ -D__SSE4_1__ -D__SSE4_2__ -D__SSE__ -D__SSSE3__ -Dtesseract_motion_planners_trajopt_ifopt_EXPORTS -IC:\ws\tesseract_wheel_shared\tesseract_ws\src\tesseract_planning\tesseract_motion_planners\trajopt_ifopt\include -IC:\ws\tesseract_wheel_shared\tesseract_ws\src\tesseract_planning\tesseract_motion_planners\core\include -IC:\ws\tesseract_wheel_shared\tesseract_ws\install\include\bullet -IC:\ws\tesseract_wheel_shared\tesseract_ws\install\include -IC:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\eigen3 -IC:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include -IC:\ws\tesseract_wheel_shared\tesseract_ws\install\include\osqp -DEIGEN_DONT_ALIGN=1 -DEIGEN_DONT_VECTORIZE=1  /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O0 /Ob0 /Zi /DNDEBUG -std:c++17 /showIncludes /Fotrajopt_ifopt\CMakeFiles\tesseract_motion_planners_trajopt_ifopt.dir\src\trajopt_ifopt_motion_planner.cpp.obj /Fdtrajopt_ifopt\CMakeFiles\tesseract_motion_planners_trajopt_ifopt.dir\ /FS -c C:\ws\tesseract_wheel_shared\tesseract_ws\src\tesseract_planning\tesseract_motion_planners\trajopt_ifopt\src\trajopt_ifopt_motion_planner.cpp
cl : Command line warning D9002 : ignoring unknown option '/O0'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/parser.h(85): warning C4251: 'YAML::Parser::m_pScanner': class 'std::unique_ptr<YAML::Scanner,std::default_delete<YAML::Scanner>>' needs to have dll-interface to be used by clients of class 'YAML::Parser'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/parser.h(85): note: see declaration of 'std::unique_ptr<YAML::Scanner,std::default_delete<YAML::Scanner>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/parser.h(86): warning C4251: 'YAML::Parser::m_pDirectives': class 'std::unique_ptr<YAML::Directives,std::default_delete<YAML::Directives>>' needs to have dll-interface to be used by clients of class 'YAML::Parser'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/parser.h(86): note: see declaration of 'std::unique_ptr<YAML::Directives,std::default_delete<YAML::Directives>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/binary.h(65): warning C4251: 'YAML::Binary::m_data': class 'std::vector<unsigned char,std::allocator<unsigned char>>' needs to have dll-interface to be used by clients of class 'YAML::Binary'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/binary.h(18): note: see declaration of 'std::vector<unsigned char,std::allocator<unsigned char>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/ostream_wrapper.h(49): warning C4251: 'YAML::ostream_wrapper::m_buffer': class 'std::vector<char,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'YAML::ostream_wrapper'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\boost/filesystem/path_traits.hpp(94): note: see declaration of 'std::vector<char,std::allocator<char>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/emitter.h(132): warning C4251: 'YAML::Emitter::m_pState': class 'std::unique_ptr<YAML::EmitterState,std::default_delete<YAML::EmitterState>>' needs to have dll-interface to be used by clients of class 'YAML::Emitter'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/emitter.h(132): note: see declaration of 'std::unique_ptr<YAML::EmitterState,std::default_delete<YAML::EmitterState>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/exceptions.h(153): warning C4275: non dll-interface class 'std::runtime_error' used as base for dll-interface class 'YAML::Exception'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\stdexcept(101): note: see declaration of 'std::runtime_error'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/exceptions.h(153): note: see declaration of 'YAML::Exception'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/exceptions.h(162): warning C4251: 'YAML::Exception::msg': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'YAML::Exception'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xstring(4871): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/node.h(135): warning C4251: 'YAML::Node::m_invalidKey': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'YAML::Node'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xstring(4871): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/node.h(136): warning C4251: 'YAML::Node::m_pMemory': class 'std::shared_ptr<YAML::detail::memory_holder>' needs to have dll-interface to be used by clients of class 'YAML::Node'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/ptr.h(24): note: see declaration of 'std::shared_ptr<YAML::detail::memory_holder>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/detail/memory.h(31): warning C4251: 'YAML::detail::memory::m_nodes': class 'std::set<YAML::detail::shared_node,std::less<YAML::detail::shared_node>,std::allocator<YAML::detail::shared_node>>' needs to have dll-interface to be used by clients of class 'YAML::detail::memory'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/detail/memory.h(30): note: see declaration of 'std::set<YAML::detail::shared_node,std::less<YAML::detail::shared_node>,std::allocator<YAML::detail::shared_node>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/detail/memory.h(42): warning C4251: 'YAML::detail::memory_holder::m_pMemory': class 'std::shared_ptr<YAML::detail::memory>' needs to have dll-interface to be used by clients of class 'YAML::detail::memory_holder'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/ptr.h(25): note: see declaration of 'std::shared_ptr<YAML::detail::memory>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/detail/node_data.h(104): warning C4251: 'YAML::detail::node_data::m_tag': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'YAML::detail::node_data'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xstring(4871): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/detail/node_data.h(108): warning C4251: 'YAML::detail::node_data::m_scalar': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'YAML::detail::node_data'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xstring(4871): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
C:\ws\tesseract_wheel_shared\vcpkg\installed\x64-windows-release\include\yaml-cpp/node/detail/node_data.h(112): warning C4251: 'YAML::detail::node_data::m_sequence': class 'std::vector<YAML::detail::node *,std::allocator<YAML::detail::node *>>' needs to have dl [... cut ...]
C:\ws\tesseract_wheel_shared\tesseract_ws\install\include\tesseract_common/type_erasure.h(197): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
C:\ws\tesseract_wheel_shared\tesseract_ws\install\include\tesseract_common/type_erasure.h(199): error C2332: 'struct': missing tag name
C:\ws\tesseract_wheel_shared\tesseract_ws\install\include\tesseract_common/type_erasure.h(199): error C2059: syntax error: 'type'
C:\ws\tesseract_wheel_shared\tesseract_ws\install\include\tesseract_common/type_erasure.h(199): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
ninja: build stopped: subcommand failed.
```

johnwason, 2022-06-30:
This one has me stumped. The compiler isn't giving much information about what is causing the problem. It is some sort of template failure.

Levi-Armstrong, 2022-06-30:
Ah, I think I have found the issue explained [here](https://github.com/skypjack/entt/issues/615). I use a interface a few places which appears it can cause issue. I bet if the following is replace with something unique `typename Interface` and `template <typename> class Instance` it may solve the issue.

Levi-Armstrong, 2022-06-30:
I would prefix them with Poly

Levi-Armstrong, 2022-06-30:
@johnwason if you leave it the same and prefix the typenames Interface and Instance with Poly in type_erasure.h does the issue go away?

Levi-Armstrong, 2022-07-02:
@johnwason can this be closed?

johnwason, 2022-07-02:
Closed by #212

closed 2022-07-02 by johnwason