tesseract-robotics/tesseract_planning #215: Passing Instructions between taskflow nodes

cpetersmeier, 2022-07-05:
Hello everyone!

## Setup
I am planning a trajectory where I want to compute the second segment `B` before the first `A`.
My Taskflow generator has three Tasks `B -> B' -> A`.

The input composite is of the format:

```
Composite
{
   Composite - A
     Start:      start
   Composite - B
     Start:      start (this gets modified during planning -> need to extract first MoveInstruction of results)
     Plan:       goal
}
```

## Question
How do I pass the first move instruction of segment `B` as the goal of segment `A`? Currently I am getting a segmentation fault error.

## Code
Since the input instructions for the `TaskInput` are const, I figured I create a subcontainer of the `TaskInput` datastructure for task A and write a taskflow node `B'` that uses `setEndInstruction()` on the subContainers' reference with the first `MoveInstruction` of task `B`.

Then the `motion_planner_task_generator` should add the end instruction to the planner requests instructions.

Here is how I construct the subcontainer. Note that I am not directly going from `B -> B'` but I added a debug printer task.

```
TaskInput A_input = input[0];

tf::Task extract_point_task = container.taskflow->emplace(
  [&](){positionalHandover(input, A_input); }
).name("Positional Handover Task");
extract_point_task.succeed(printer_task);
```

Here is my code of `B'`:

```

void CustomTaskflow::positionalHandover(TaskInput& input, TaskInput& A_input){
  CompositeInstruction& results = input.getResults()->as<CompositeInstruction>();  // Segmentation fault occurs here in isCompositeInstruction() -> getType()

  CompositeInstruction& B_results = results.at(1).as<CompositeInstruction>();

  // get the first move instruction from this path
  MoveInstruction& first_move_instruction = *getFirstMoveInstruction(B_results);
  A_input.setEndInstruction(first_move_instruction);
  
  return;
}
```

I tried using the latest taskflow v3.4, which supports passing additional data into the tasks but hat no luck with that either.

```
Debug:   worker 0 ready to run Positional Handover Task
[app_node-1] process has died [pid 113693, exit code -11 ... (log files, etc ..)]
```

cpetersmeier, 2022-07-05:
Actually, the segmentation fault can be mitigated by capturing the `input` variable by value 

```
[input, &A_input](){positionalHandover(input, A_input); }
```

and just using a reference for the `A_input` subcontainer.

```
void CustomTaskflow::positionalHandover(TaskInput input, TaskInput& A_input){
```

Levi-Armstrong, 2022-07-05:
You will need to write your own graph taskflow. An example of this can be seen by looking at the raster graph taskflow.

cpetersmeier, 2022-07-05:
Thanks for the quick reply! The code above is contained in my own graph taskflow. I looked at this exampe:

https://github.com/tesseract-robotics/tesseract_planning/blob/86ac76994c8713b318c6a0679878f9a19a16cf17/tesseract_process_managers/src/taskflow_generators/raster_only_global_taskflow.cpp#L149-L168

However when I do a `setEndInstruction()` in my positionalHandover task and try to read it out when I pass this modified `TaskInput` to the next motion planner graph node, I get a `NullInstruction`. 

I think the example modifies the seed of the next planner? Would this accomplish the same effect as adding an end instruction which gets then added to the input instructions here?

https://github.com/tesseract-robotics/tesseract_planning/blob/86ac76994c8713b318c6a0679878f9a19a16cf17/tesseract_process_managers/src/task_generators/motion_planner_task_generator.cpp#L137

Levi-Armstrong, 2022-07-05:
Take a look at this section of code. You want to provide it with the index instead of the actual waypoint. There is a comment explaining why this has to be done, but in short is because the result data structure has not been generated so you are unable to reference it directly.

https://github.com/tesseract-robotics/tesseract_planning/blob/86ac76994c8713b318c6a0679878f9a19a16cf17/tesseract_process_managers/src/taskflow_generators/raster_taskflow.cpp#L118-L143

Levi-Armstrong, 2022-07-05:
This is the index into the vector of the high level composite instruction. In your case I believe the this should be index 1 for B.

cpetersmeier, 2022-07-06:
Awesome, thanks a lot for your help and quick response!

If I may ask: Is there a way to let the rviz trajectory visualization plugin display the full path? Currently it only loops through the `B` section. 

I also flattened out all `MoveInstructions` of the result composite by default for the toolpath extraction, otherwise I would not get a path displayed in rviz.