tesseract-robotics/tesseract_planning #229: build errr:ws_tesseract/src/tesseract_planning-0.12.0/tesseract_motion_planners/core/src/simple/simple_motion_planner.cpp:146:89: error: no matching function for call to 'satisfiesPositionLimits(Eigen::VectorXd&, Eigen::MatrixX2d)'   146 |     assert(tesseract_common::satisfiesPositionLimits(jp, manip->getLimits().joint_limits));

dbdxnuliba, 2022-07-26:
ws_tesseract/src/tesseract_planning-0.12.0/tesseract_motion_planners/core/src/simple/simple_motion_planner.cpp:146:89: error: no matching function for call to 'satisfiesPositionLimits(Eigen::VectorXd&, Eigen::MatrixX2d)'
  146 |     assert(tesseract_common::satisfiesPositionLimits(jp, manip->getLimits().joint_limits));

dbdxnuliba, 2022-07-26:
I use tesseract_planning-0.12.0 and tesseract-0.13.0

Levi-Armstrong, 2022-07-26:
Use the following tags. The next release coming up will set all tags to the same to make it easier to know which tags to use across the tesseract repositories.

- tesseract: 0.10.0
- tesseract_planning: 0.12.0
- tesseract_qt: 0.2.0
- tesseract_ros: 0.5.1
- trajopt: 0.3.0

dbdxnuliba, 2022-07-27:
thanks a lot!

Levi-Armstrong, 2022-08-01:
@dbdxnuliba  Can this be closed?

closed 2023-02-26 by Levi-Armstrong