tesseract-robotics/tesseract_planning #244: unable to build tesseract_command_language due to isNull not available as member function of some structs

thomasw2, 2022-09-08:
We tried to build some tesseract repositories in a catkin workspace. When building, we encountered an error:
[image]

we had the following repos in src during building:
descartes_light, ifopt, osqp, ros_industrial_cmake_boilerplate, tesseract, tesseract_planning, tesseract_ros, trajopt
The tesseract repos were all on master branch. 
Our system is Ubuntu 20.04, running ROS Noetic. We cloned the above repos and ran catkin build.
Are there any suggestions for how to get around this issue?

Levi-Armstrong, 2022-09-08:
Can you provide the tags/branches for each of the repositories? The tesseract repositories I recommend using the latest tag 0.13.X where the X may be different on each repository.

Levi-Armstrong, 2022-09-09:
Can you check the latest tag on tesseract, tesseract_planning, tesseract_qt and tesseract_ros. It should be 0.13.X?

Levi-Armstrong, 2023-02-28:
Reopen if still an issue

closed 2023-02-28 by Levi-Armstrong