tesseract-robotics/tesseract_planning #254: Add time parameterization interface and optional implementation builds

marip8, 2022-10-10:
It seems like all 3 existing implementations of time parameterization are intended to perform the same function. We should create an abstract interface for time parameterization and add the ability to optionally disable builds of implementations that need additional 3rd party dependencies (e.g., Ruckig)

Levi-Armstrong, 2022-10-11:
I agree. We can still make them optional to build now if that is something needed now.

marip8, 2023-11-06:
Optional builds added in #373

Levi-Armstrong, 2023-11-07:
@marip8 Are you working on the interface?

Levi-Armstrong, 2023-11-07:
Can we make the interface simpler so these are not part of the interface but just part of the yaml config/profiles?

marip8, 2023-11-07:
> Can we make the interface simpler so these are not part of the interface but just part of the yaml config/profiles?

Depends. We could do this suggestion by just moving the algorithms into their respective tasks in task composer (like in [SNP with the constant TCP speed time paramterization classes](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/tree/master/snp_motion_planning/src/plugins/tasks). I guess the question is whether or not there is value in having a stand-alone time parameterization class that exists outside of task composer. If there is value there, then it would make sense to create a common interface and force all implementations to provide that interface.

I personally only have only used the time parameterization algorithms as a part of a task pipeline, so I might lean towards dissolving `tesseract_time_parameterization` into `tesseract_task_composer` and creating libraries for the algorithms themselves in case anyone wants to use want for a special use-case

closed 2025-01-02 by Levi-Armstrong