tesseract-robotics/tesseract_planning #259: [0.14.0] rosdep install error for tesseract_task_composer

seanmc602, 2022-11-02:
Output from Tesseract portion of a Docker build. The previous version of tesseract_ros, 0.6.0, installs deps fine.

```
Step 28/38 : RUN mkdir -p $TESSERACT_WS_TO_SOURCE && cd $TESSERACT_WS_TO_SOURCE &&     git clone -b 0.14.0 https://github.com/tesseract-robotics/tesseract_ros.git
 ---> Running in 332dc163f63c
Cloning into 'tesseract_ros'...
Note: switching to '0c3d1899d35a045cd48f1202e85bf222a466fd62'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Removing intermediate container 332dc163f63c
 ---> d3a17b5c1950
Step 29/38 : RUN wstool init $TESSERACT_WS_TO_SOURCE $TESSERACT_WS_TO_SOURCE/tesseract_ros/dependencies.rosinstall
 ---> Running in cd38e7dfdcf7
Using initial elements from: /tesseract_ws/src/tesseract_ros/dependencies.rosinstall
Writing /tesseract_ws/src/.rosinstall
Cloning into '/tesseract_ws/src/ros_industrial_cmake_boilerplate'...
Cloning into '/tesseract_ws/src/tesseract'...
Cloning into '/tesseract_ws/src/trajopt'...
Cloning into '/tesseract_ws/src/tesseract_planning'...
Cloning into '/tesseract_ws/src/tesseract_qt'...
Cloning into '/tesseract_ws/src/descartes_light'...
Cloning into '/tesseract_ws/src/opw_kinematics'...
Cloning into '/tesseract_ws/src/ifopt'...
[ros_industrial_cmake_boilerplate] Fetching https://github.com/ros-industrial/ros_industrial_cmake_boilerplate.git (version 0.3.1) to /tesseract_ws/src/ros_industrial_cmake_boilerplate
[ros_industrial_cmake_boilerplate] Done.
[tesseract] Fetching https://github.com/tesseract-robotics/tesseract.git (version 0.13.1) to /tesseract_ws/src/tesseract
[tesseract] Done.
[trajopt] Fetching https://github.com/tesseract-robotics/trajopt.git (version 0.3.0) to /tesseract_ws/src/trajopt
[trajopt] Done.
[tesseract_planning] Fetching https://github.com/tesseract-robotics/tesseract_planning.git (version 0.13.0) to /tesseract_ws/src/tesseract_planning
[tesseract_planning] Done.
[tesseract_qt] Fetching https://github.com/tesseract-robotics/tesseract_qt.git (version 0.13.0) to /tesseract_ws/src/tesseract_qt
[tesseract_qt] Done.
[descartes_light] Fetching https://github.com/swri-robotics/descartes_light.git (version 0.3.0) to /tesseract_ws/src/descartes_light
[descartes_light] Done.
[opw_kinematics] Fetching https://github.com/Jmeyer1292/opw_kinematics.git (version 0.4.5) to /tesseract_ws/src/opw_kinematics
[opw_kinematics] Done.
[ifopt] Fetching https://github.com/ethz-adrl/ifopt.git (version 2.1.3) to /tesseract_ws/src/ifopt
[ifopt] Done.

update complete.
Removing intermediate container cd38e7dfdcf7
 ---> 63dbfeb6f320
Step 30/38 : RUN rosdep init && rosdep update --rosdistro ${ROS_DISTRO}
 ---> Running in 6d90c57dd1c4
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip distro "ardent" different from requested "noetic"
Skip distro "bouncy" different from requested "noetic"
Skip distro "crystal" different from requested "noetic"
Skip distro "dashing" different from requested "noetic"
Skip distro "eloquent" different from requested "noetic"
Skip distro "foxy" different from requested "noetic"
Skip distro "galactic" different from requested "noetic"
Skip distro "groovy" different from requested "noetic"
Skip distro "humble" different from requested "noetic"
Skip distro "hydro" different from requested "noetic"
Skip distro "indigo" different from requested "noetic"
Skip distro "jade" different from requested "noetic"
Skip distro "kinetic" different from requested "noetic"
Skip distro "lunar" different from requested "noetic"
Skip distro "melodic" different from requested "noetic"
Add distro "noetic"
Skip distro "rolling" different from requested "noetic"
updated cache in /root/.ros/rosdep/sources.cache
Removing intermediate container 6d90c57dd1c4
 ---> 36b70946dd25
Step 31/38 : RUN bash -c "source /opt/ros/noetic/setup.bash"     && rosdep install -q --from-paths $TESSERACT_WS_TO_SOURCE --ignore-src -y --rosdistro ${ROS_DISTRO}
 ---> Running in 2c98fe7d24f1
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tesseract_rosutils: Cannot locate rosdep definition for [tesseract_task_composer]
tesseract_planning_server: Cannot locate rosdep definition for [tesseract_task_composer]
```

Levi-Armstrong, 2022-11-02:
Yea, sorry, the release tag actually does not have the updated rosinstall files. I will recreate a new tag which has them include.

rjoomen, 2023-03-31:
I suppose this can be closed?

closed 2023-03-31 by Levi-Armstrong