tesseract-robotics/tesseract_planning #260: [0.14.0] Profile not found in namespace 'SIMPLE_PLANNER'

miranisaurabh, 2022-11-04:
I am using version 0.14.0 and am facing the issue of SimplePlanner profile not being found even though I am explicitly adding the profile to the dictionary. Since I am using separate waypoint profile ("1" in code below) and waypoint path profile ("path_1"), I need to add the SimplePlanner Profile explicitly. I also checked before and after adding, the dictionary does have it, it's just not found by the planner. I was using 0.12.0 before this and adding SimplePlanner profile in a similar way worked perfectly fine. I am also adding TrajOptPlanProfile profile for "path_1" but it doesn't complain for that.

This is how I am adding SimplePlanner profile:
```
std::cout << "hasProfile SimplePlannerPlanProfile before: "
          << profileDictPtr->hasProfile<SimplePlannerPlanProfile>(
                 profile_ns::SIMPLE_DEFAULT_NAMESPACE, "path_1")
          << std::endl;

profileDictPtr->addProfile<SimplePlannerPlanProfile>(
    profile_ns::SIMPLE_DEFAULT_NAMESPACE,
    "path_1",
    std::make_shared<SimplePlannerLVSPlanProfile>());

std::cout << "hasProfile SimplePlannerPlanProfile after: "
          << profileDictPtr->hasProfile<SimplePlannerPlanProfile>(
                 profile_ns::SIMPLE_DEFAULT_NAMESPACE, "path_1")
          << std::endl;

```

I add the waypoints using:
```
CartesianWaypointPoly waypoint{CartesianWaypoint(robotPose)};
MoveInstruction waypointInstruction(waypoint, MoveInstructionType::LINEAR, "1", "path_1");
waypointInstruction.setDescription("Waypoint");
CompositeInstructionUPtr->appendMoveInstruction(waypointInstruction)
```

Finally I start the planner using:
```
TaskComposerDataStorage inputData;
inputData.setData("input_program", *m_robotProgram);
TaskComposerProblem problem(m_environment, inputData);
CompositeInstructionUPtr->print("Program: ");
TaskComposerInput input(problem, profileDictPtr);
TrajOptMotionPipelineTask task("input_program", "output_program");
TaskComposerFuture::UPtr taskFuture = TaskflowTaskComposerExecutorUPtr->run(task, input);
taskFuture->wait();
```
I enabled the DEBUG mode and got the following output:
```
hasProfile SimplePlannerPlanProfile before: 0
hasProfile SimplePlannerPlanProfile after: 1
Debug:   Environment, getGroupJointNames(manipulator) cache miss!
Program: Composite Instruction, Description: Tesseract Composite Instruction
Program: --- Start Instruction, Description: Tesseract Move Instruction
Program: {
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.7, -0.2, 0.005
, Description: Waypoint
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.7, 0.2, 0.005
, Description: Waypoint
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.45, 0.2, 0.005
, Description: Waypoint
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.45, -0.2, 0.005
, Description: Waypoint
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.7, -0.2, 0.005
, Description: Waypoint
Program: }
Debug:   Environment, getJointGroup(manipulator) cache miss!
Debug:   Environment, getGroupJointNames(manipulator) cache hit!
Debug:   Profile 'path_1' was not found in namespace 'SIMPLE_PLANNER' for type 'N18tesseract_planning24SimplePlannerPlanProfileE'. Using default if available. Available profiles:
Debug:   1
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Profile 'path_1' was not found in namespace 'SIMPLE_PLANNER' for type 'N18tesseract_planning24SimplePlannerPlanProfileE'. Using default if available. Available profiles:
Debug:   1
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Profile 'path_1' was not found in namespace 'SIMPLE_PLANNER' for type 'N18tesseract_planning24SimplePlannerPlanProfileE'. Using default if available. Available profiles:
Debug:   1
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Profile 'path_1' was not found in namespace 'SIMPLE_PLANNER' for type 'N18tesseract_planning24SimplePlannerPlanProfileE'. Using default if available. Available profiles:
Debug:   1
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Profile 'path_1' was not found in namespace 'SIMPLE_PLANNER' for type 'N18tesseract_planning24SimplePlannerPlanProfileE'. Using default if available. Available profiles:
Debug:   1
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Seed Min Length Task Succeeded!
Debug:   Environment, getKinematicGroup(manipulator, ) cache miss!
Debug:   Environment, getGroupJointNames(manipulator) cache hit!
[INFO] iteration 1
---TrajOpt Table here
```
As you can see it says profile wasn't found even thought it's in the dictionary.
I tried narrowing down the error but couldn't locate it and hence need your help in figuring this out. Thank you.

Levi-Armstrong, 2022-11-09:
Thanks for reporting. I will look into this and report back.

marip8, 2022-12-28:
I made quite a few changes and improvements to the way profiles are added to and retrieved from the profile dictionary to the `feat/ProfileRefactor` branch in #255. The profile dictionary no longer returns default profiles when specified profiles cannot be found, and the error reporting is also improved substantially. I've had a lot of issues in the past with the planning server not finding my specific profiles and also with profile names not being specified correctly in all the right places. These changes have helped me a lot with those problems. If you're building `tesseract` from source, you might try out that branch to see if it solves your problem. I think it's one commit on top of 0.14.0

Levi-Armstrong, 2023-01-23:
@miranisaurabh I have recently fixed some issues which may resolve the issue you were having.

miranisaurabh, 2023-01-31:
Thanks for the update! I'll try again and will let you know if I still face this issue.

rjoomen, 2023-03-09:
The tesseract_ros examples also produce a lot of errors/warnings regarding profiles that can't be found. The [feat/ProfileRefactor](https://github.com/tesseract-robotics/tesseract_planning/tree/feat/ProfileRefactor) branch is by now somewhat behind `master`, so I'm not sure I can meaningfully test these improvements.

@marip8, what's the status on this?

marip8, 2023-03-09:
I think we basically need to open #255 against master and merge. I think those commits should still sit easily on top of the current master branch. I'll try to open that PR today

rjoomen, 2023-08-22:
@marip8, friendly ping. Did you manage to work on this?

marip8, 2023-08-22:
See #367; just needs a little more work

closed 2025-04-23 by rjoomen