tesseract-robotics/tesseract_planning #275: OMPL input-output lengths linked

marrts, 2023-02-06:
Currently the OMPL planner expects to return the same length of a trajectory that it got in. It should be able to return any length, especially now as we've moved to the TaskComposer setup.

https://github.com/tesseract-robotics/tesseract_planning/blob/b171caf1b1e0cc4f8f34e3e989b466e825ab14a9/tesseract_motion_planners/ompl/src/ompl_motion_planner.cpp#L225-L228

Levi-Armstrong, 2023-02-06:
That is correct. I recommend adding an option to the profile to allow disabling this functionality.

marrts, 2023-02-06:
Would it make more sense to never copy the input and just create a new CI every time? And then add optional parameter(s) for min and/or max length?

marip8, 2023-02-06:
FWIW, I think the output is being copied from the input currently to retain all of the meta-data (i.e., manipulator info, profile names, etc.). As long as we capture that data correctly, it would be okay to build a new CI from scratch

Levi-Armstrong, 2025-01-04:
I do not think this is still valid with recent changes to OMPL. Please reopen if that is not the case.

closed 2025-01-04 by Levi-Armstrong