tesseract-robotics/tesseract_planning #276: Cartesian Motion Planning Pipeline Hardcoded Tasks

marrts, 2023-02-06:
Since the update to using TaskComposer all the taskflow processes have been updated. Previously one could construct a raster taskflow by passing in custom freespace and raster planners, but now all the processes are hardcoded to use the default implementations. 

See example here:
https://github.com/tesseract-robotics/tesseract_planning/blob/b171caf1b1e0cc4f8f34e3e989b466e825ab14a9/tesseract_task_composer/src/nodes/raster_ct_motion_task.cpp#L205

Levi-Armstrong, 2023-02-06:
I recently ran into this. I am going to switch these classes to be a template to take these types to allow changing these sub tasks.

marrts, 2023-02-09:
I made a custom implementation of the raster and global planners [here](https://github.com/marrts/scan_n_plan_workshop/blob/834da7141782eb49fcd5a556821ed471c7d04afc/snp_motion_planning/src/raster_task/snp_raster_motion_task.cpp) that takes in functions that generate unique pointers to task node objects. They default as nullptrs and if they aren't set they use the Tesseract defaults, otherwise they use a custom one. I couldn't figure out how to do this easily with templates, but I found this method worked pretty well for me.

Levi-Armstrong, 2023-02-15:
This should be addressed by PR #279

Levi-Armstrong, 2023-02-26:
Addressed by PR #282

closed 2023-02-26 by Levi-Armstrong