tesseract-robotics/tesseract_planning #277: OMPL Planner with Seed

marrts, 2023-02-06:
Since updating to using the TaskComposer structure the CompsiteInstruction was reworked so that now waypoints have seeds rather than the entire Instruction. This is a problem when trying to use OMPL because now when it sees an interpolated input it actually sees a series of MoveInstructions and it therefore creates a series of very small OMPL problems that must hit every waypoint. This problem can be observed by running the default `FreespaceMotionPipelineTask`. May be related to #275,

Levi-Armstrong, 2023-02-06:
This should not be happening. Were you by chance using one of the global taskflows? I ran into this here recently and fixed it in a recently merged PR causing this type of behavior.

Levi-Armstrong, 2023-02-06:
I was seeing this behavior which was a result of the global descartes task not formatting the results for input. This [PR](https://github.com/tesseract-robotics/tesseract_planning/pull/271) fixes the issue.

marrts, 2023-02-06:
This was with a previous global task feeding the rest of the process, I will try the latest version

marrts, 2023-02-06:
Looks like that did resolve the OMPL issue, but now I'm running into more issues with ISP and TrajOpt not having enough waypoints to run. Seems like OMPL is just outputting a CI of only 2 waypoints. Need to investigate it further.

Levi-Armstrong, 2023-02-07:
What pipeline are you using?

marrts, 2023-02-07:
I'm using `RasterCtMotionTask`. Here is an example failure:
```
[snp_motion_planning_node-7] Debug:   Seed Min Length Task Succeeded!
[snp_motion_planning_node-7] Debug:   Environment, getKinematicGroup(manipulator, ) cache hit!
[snp_motion_planning_node-7] Debug:   Environment, getKinematicGroup(manipulator, ) cache hit!
[snp_motion_planning_node-7] Debug:   Motion Planner process succeeded
[snp_motion_planning_node-7] Debug:   Environment, getKinematicGroup(manipulator, ) cache hit!
[snp_motion_planning_node-7] Debug:   Profile 'SNPD' was not found in namespace 'TRAJOPT' for type 'N18tesseract_planning20TrajOptSolverProfileE'. Using default if available. Available profiles:
[snp_motion_planning_node-7] [INFO] iteration 1
[snp_motion_planning_node-7] Debug:   Seed Min Length Task Succeeded!
[snp_motion_planning_node-7] Debug:   Environment, getKinematicGroup(manipulator, ) cache hit!
[snp_motion_planning_node-7] [INFO]  
[snp_motion_planning_node-7] 
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] |                               ROS Industrial                                |
[snp_motion_planning_node-7] |                           TrajOpt Motion Planning                           |
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] |      merit |   oldexact |  new_exact |    dapprox |     dexact |      ratio | -                
[snp_motion_planning_node-7] | --------------------------------------------------------------------------- | COSTS
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] | ---------- |  0.000e+00 |  0.000e+00 |   ------   |   ------   |   ------   | SUM COSTS 
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] | --------------------------------------------------------------------------- | CONSTRAINTS
[snp_motion_planning_node-7] |  1.000e+01 |  0.000e+00 |  2.465e-30 | -8.882e-15 | -2.465e-30 |   ------   | joint_waypoint_0 
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] | ---------- |  0.000e+00 |  2.465e-31 |   ------   |   ------   |   ------   | SUM CONSTRAINTS (WITHOUT MERIT) 
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] | ---------- |  0.000e+00 |  2.465e-30 | -8.882e-15 | -2.465e-30 |  2.776e-16 | TOTAL = SUM COSTS + SUM CONSTRAINTS (WITH MERIT)
[snp_motion_planning_node-7] | =========================================================================== |
[snp_motion_planning_node-7] [INFO] converged because improvement was small (-8.882e-15 < 1.000e-04)
[snp_motion_planning_node-7] [INFO] woo-hoo! constraints are satisfied (to tolerance 1.00e-04)
[snp_motion_planning_node-7] [INFO] 
[snp_motion_planning_node-7] ==================
[snp_motion_planning_node-7] Optimization results:
[snp_motion_planning_node-7] status: CONVERGED
[snp_motion_planning_node-7] cost values: ()
[snp_motion_planning_node-7] constraint violations: (0)
[snp_motion_planning_node-7] n func evals: 2
[snp_motion_planning_node-7] n qp solves: 1
[snp_motion_planning_node-7] ==================
[snp_motion_planning_node-7] Debug:   Motion Planner process succeeded
[snp_motion_planning_node-7] Debug:   Profile 'SNPD' was not found in namespace 'DiscreteContactCheckTask' for type 'N18tesseract_planning19ContactCheckProfileE'. Using default if available. Available profiles:
[snp_motion_planning_node-7] Debug:   Environment, getJointGroup(manipulator) cache hit!
[snp_motion_planning_node-7] Debug:   Discrete contact check succeeded
[snp_motion_planning_node-7] Debug:   Environment, getJointGroup(manipulator) cache hit!
[snp_motion_planning_node-7] Error:   iterative_spline_parameterization: number of waypoints 1, needs to be greater than 3.
[snp_motion_planning_node-7]          at line 365 in /home/ros-industrial/ros2_workspaces/scan_n_plan_ws/src/tesseract_planning/tesseract_time_parameterization/isp/src/iterative_spline_parameterization.cpp
[snp_motion_planning_node-7] Info:    Failed to perform iterative spline time parameterization for process input: from_start
```

It's also likely that I am setting up the start wrong since the removal of the start instruction. Here is where I am trying to do this:

https://github.com/marrts/scan_n_plan_workshop/blob/d4b794dc85afbbf88b7597a05115ac5b0c55cb50/snp_motion_planning/src/planning_server.cpp#L172-L208

Also here is the xdot graph for reference:  [image]

Levi-Armstrong, 2023-02-08:
I have not used the ct version yet but I know the ft version works. What happens if you try that pipeline?

closed 2023-02-08 by marrts