tesseract-robotics/tesseract_planning #280: Building Error for Ruckig

zhiyong-009, 2023-02-11:
When building on ROS 2, encountered the following error, how do i resolve this?

  CMakeLists.txt:30 (initialize_code_coverage)

Building ISP
Building TOTG
Building Ruckig
CMake Error at ruckig/CMakeLists.txt:1 (find_package):
  By not providing "Findruckig.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ruckig", but
  CMake did not find one.

  Could not find a package configuration file provided by "ruckig" with any
  of the following names:

    ruckigConfig.cmake
    ruckig-config.cmake

  Add the installation prefix of "ruckig" to CMAKE_PREFIX_PATH or set
  "ruckig_DIR" to a directory containing one of the above files.  If "ruckig"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< tesseract_time_parameterization [0.55s, exited with code 1]

zhiyong-009, 2023-02-11:
Error occured when building the tesseract_planning package

Levi-Armstrong, 2023-02-11:
You need to install rucking or disable building it's feature. Check the time parameterization packages CMake file for the name of the parameter.

closed 2023-02-12 by zhiyong-009