tesseract-robotics/tesseract_planning #301: Task Composer Configuration File Loading

marip8, 2023-04-05:
Is the following function intended to load the task composer config from a string filename or a YAML string?

https://github.com/tesseract-robotics/tesseract_planning/blob/e92083398c11da5a808794bed82cec392f2c4625/tesseract_task_composer/src/task_composer_plugin_factory.cpp#L89

If the intent is to load from a string filename (as in most applications I have encountered so far), then this function needs to use `YAML::LoadFile` instead of `YAML::Load`

Levi-Armstrong, 2023-04-05:
The header documentation state that it is loading it as a string. The method above states loading from file using `tesseract_common::fs::path`.

Levi-Armstrong, 2023-04-05:
Probably good to update the header to state yaml string to avoid confusion.

Levi-Armstrong, 2023-06-29:
I believe this can be closed. Reopen if not the case.

closed 2023-06-29 by Levi-Armstrong