tesseract-robotics/tesseract_planning #314: Output Key Naming Scheme in Raster Motion Task

marrts, 2023-05-09:
Perhaps I am doing something wrong, but there appears to be an issue with how the output keys are named in the [raster motion task](https://github.com/tesseract-robotics/tesseract_planning/blob/82e7316a2dff4c2fb45801eb0659ac3c07c2217c/tesseract_task_composer/src/nodes/raster_motion_task.cpp). Whenever I run the motion planning everything seems to plan properly (I can see rasters, transitions, and freespace plans occurring in the terminal), but the resulting composite produces a trajectory that just has a single transition and a single freespace move each repeated a few times. My current state of the scan_n_plan workshop where I am testing this  can be found [here](https://github.com/marrts/scan_n_plan_workshop/tree/update/tesseract_0.16). 

When I attach a debugger and stop it [here](https://github.com/tesseract-robotics/tesseract_planning/blob/82e7316a2dff4c2fb45801eb0659ac3c07c2217c/tesseract_task_composer/src/nodes/raster_motion_task.cpp#L445) you can see the output data for the rasters and transitions have the same keys: 
[image]

I think whenever the `input.data_storage.getData(...)` is called it will access the wrong memory some times due to this repeated naming scheme.

The other option is potentially there is a way to fix this in the plugins file used for generating the taskflow. You can find the one I am using [here](https://github.com/marrts/scan_n_plan_workshop/blob/7d1c0bd0210c6cdf3f830ef9f658e3c42dda69b6/snp_motion_planning/config/task_composer_plugins.yaml), or view a relevant snippet here:
```yaml
           RasterMotionTask:
              class: RasterMotionTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
                freespace:
                  task: SNPFreespacePipeline
                  input_remapping:
                    input_data: output_data
                  input_indexing: [output_data]
                  output_indexing: [output_data]
                raster:
                  task: SNPCartesianPipeline
                  input_remapping:
                    input_data: output_data
                  input_indexing: [output_data]
                  output_indexing: [output_data]
                transition:
                  task: SNPTransitionPipeline
                  input_remapping:
                    input_data: output_data
                  input_indexing: [output_data]
                  output_indexing: [output_data]
```

Here is a video of the bad trajectory:

https://github.com/tesseract-robotics/tesseract_planning/assets/41449746/d1cf15b6-ba34-4a38-afea-02e14fe37714

Levi-Armstrong, 2023-05-09:
At some point I need to write some unit tests. I think possible that they need different names, but I will have to create a unit tests to make sure. May try the one below, but can you also share the yam for each of the tasks being used. This is need to properly define the remapping info?

``` yaml
           RasterMotionTask:
              class: RasterMotionTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
                freespace:
                  task: SNPFreespacePipeline
                  input_remapping:
                    input_data: freespace_output_data
                  output_remapping:
                    output_data: freespace_output_data
                  input_indexing: [freespace_output_data]
                  output_indexing: [freespace_output_data]
                raster:
                  task: SNPCartesianPipeline
                  input_remapping:
                    input_data: raster_output_data
                  output_remapping:
                    output_data: raster_output_data
                  input_indexing: [raster_output_data]
                  output_indexing: [raster_output_data]
                transition:
                  task: SNPTransitionPipeline
                  input_remapping:
                    input_data: transition_output_data
                  output_remapping:
                    output_data: transition_output_data
                  input_indexing: [transition_output_data]
                  output_indexing: [transition_output_data]
```

marrts, 2023-05-09:
The full yaml file is linked above, but you can also see it here: https://github.com/marrts/scan_n_plan_workshop/blob/7d1c0bd0210c6cdf3f830ef9f658e3c42dda69b6/snp_motion_planning/config/task_composer_plugins.yaml.

The changes you suggested did not seem to work on my first attempt, It seems to not be following the exact taskflow as I want it to.

marrts, 2023-06-15:
Closing as this was resolved with #324.

closed 2023-06-15 by marrts