tesseract-robotics/tesseract_planning #321: How to get control of the number of the trajectory points?

decdox, 2023-05-23:
HI!
I am beginning to use Tesseract for my motion planning, and I have looked through all the examples given in repo [tesseract_ros_examples](https://github.com/tesseract-robotics/tesseract_ros/tree/master/tesseract_ros_examples/src). All planning results in this repo only provide a few trajectory points: 
```
basic_cartesian_example.launch  13 points
car_seat_example.launch  7 + 7 
freespace_hybrid_example.launch  8
freespace_ompl_example.launch  8
glass_upright_example.launch 7
pick_and_place_example.launch  11 + 13
puzzle_piece_auxillary_axes_example.launch  436 + 436
puzzle_piece_example.launch  436 + 436
```
Except the `puzzle_piece_example` which has over 400 points, I think this is because there are many cartesian waypoint added from `package://tesseract_support/urdf/puzzle_bent.csv`. Other planning results provide no more than 15 points. 

I also notice that in this [issue](https://github.com/tesseract-robotics/tesseract_planning/issues/101), the figures show many waypoints although the user only defined two instruction waypoints by
```
Waypoint wp0 = CartesianWaypoint(getPoses()[0]);
Waypoint wp1 = CartesianWaypoint(getPoses()[1]);

PlanInstruction plan_f0(wp0, PlanInstructionType::LINEAR, "FREESPACE");
PlanInstruction plan_f1(wp1, PlanInstructionType::LINEAR, "RASTER");
program.push_back(plan_f0);
program.push_back(plan_f1);
```

I am wonderring **if there is any method to control the number of the waypoints**? Thank you anyway.

decdox, 2023-05-24:
Thank you for quick reply. 
Could you please give more details about this, meaning how to implement in the code? I found that no examples given in the tesseract_examples are implemented with the `SimplePlanner` but others like `TrajOptDefaultPlannerProfile`, `OMPLDefaultPlanProfile` and `TrajOptIfoptDefaultPlanProfile`. I have tried the `TrajOptDefaultPlannerProfile` and the `TrajOptIfoptDefaultPlanProfile`, they both generate only 7 trajectory waypoints.

Part of my codes are below:
```
    std::cout << "========== TrajOpt ==========" << std::endl;
    // composite profile
    auto composite_profile = std::make_shared<TrajOptDefaultCompositeProfile>();
    composite_profile->collision_cost_config.enabled = true;
    composite_profile->collision_cost_config.type = trajopt::CollisionEvaluatorType::DISCRETE_CONTINUOUS;
    composite_profile->collision_cost_config.safety_margin = 0.01;
    composite_profile->collision_cost_config.safety_margin_buffer = 0.01;
    composite_profile->collision_cost_config.coeff = 1;
    composite_profile->collision_constraint_config.enabled = true;
    composite_profile->collision_constraint_config.type = trajopt::CollisionEvaluatorType::DISCRETE_CONTINUOUS;
    composite_profile->collision_constraint_config.safety_margin = 0.01;
    composite_profile->collision_constraint_config.safety_margin_buffer = 0.01;
    composite_profile->collision_constraint_config.coeff = 1;
    composite_profile->smooth_velocities = true;
    composite_profile->smooth_accelerations = false;
    composite_profile->smooth_jerks = false;
    composite_profile->velocity_coeff = Eigen::VectorXd::Ones(1);
    profiles->addProfile<TrajOptCompositeProfile>(TRAJOPT_DEFAULT_NAMESPACE, "UPRIGHT", composite_profile);
    
    // plan profile
    auto plan_profile = std::make_shared<TrajOptDefaultPlanProfile>();
    plan_profile->cartesian_coeff = Eigen::VectorXd::Constant(6, 1, 5);
    // plan_profile->cartesian_coeff = Eigen::VectorXd::Ones(6);
    plan_profile->cartesian_coeff(0) = 0;
    plan_profile->cartesian_coeff(1) = 0;
    plan_profile->cartesian_coeff(2) = 0;
    plan_profile->joint_coeff = Eigen::VectorXd::Ones(7);

    // Add profile to Dictionary
    profiles->addProfile<TrajOptPlanProfile>(TRAJOPT_DEFAULT_NAMESPACE, "UPRIGHT", plan_profile);
```
```
    std::cout << "========== TrajOptIfopt ==========" << std::endl;
    auto composite_profile = std::make_shared<TrajOptIfoptDefaultCompositeProfile>();
    // composite_profile->collision_cost_config->type = trajopt::CollisionEvaluatorType::DISCRETE_CONTINUOUS;
    composite_profile->collision_cost_config->collision_margin_buffer = 0.01;
    composite_profile->smooth_velocities = true;
    composite_profile->smooth_accelerations = false;
    composite_profile->smooth_jerks = false;
    composite_profile->velocity_coeff = Eigen::VectorXd::Ones(1);
    profiles->addProfile<TrajOptIfoptCompositeProfile>(TRAJOPT_DEFAULT_NAMESPACE, "UPRIGHT", composite_profile);

    auto plan_profile = std::make_shared<TrajOptIfoptDefaultPlanProfile>();
    plan_profile->cartesian_coeff = Eigen::VectorXd::Constant(6, 1, 5);
    plan_profile->cartesian_coeff(0) = 0;
    plan_profile->cartesian_coeff(1) = 0;
    plan_profile->cartesian_coeff(2) = 0;    
    profiles->addProfile<TrajOptIfoptPlanProfile>(TRAJOPT_DEFAULT_NAMESPACE, "UPRIGHT", plan_profile);
```

BTW, there are two things need to help.
- Interpolation in `SimplePlanner`
`[marip8](https://github.com/marip8)` last year replied in this [issue](https://github.com/tesseract-robotics/tesseract_planning/issues/101#issuecomment-1139694197) that 

> The default SimplePlanner (which is a part of most planning taskflows, i.e. process planners) performs interpolation based on these strategies using a "longest-valid-segment" technique, **meaning it adds an interpolated point a some discrete Cartesian or joint distance (e.g. every 25mm or every 1 degree of joint space movement).** That's why your video shows many Cartesian waypoints when you only actually gave the planner 2 waypoints. There are other SimplePlanner profiles that have different behavior, such as copying the first waypoint n times, interpolating over n states, etc.

How to switch to interpolate point every 25mm mentioned above.

- Does the current version of `tesseract` support `term_info`
I found that in the earlier version of `tesseract`, one can add  `TermInfo` like `JointVelTermInfo` to set joint position limit, velocity limit, acceleration limit, etc,. But currently it seem not. Is there any method to add such limit or term info?

Thank you again!

Levi-Armstrong, 2023-05-27:
You want to use one of the lvs profiles located [here](https://github.com/tesseract-robotics/tesseract_planning/tree/master/tesseract_motion_planners/core/include/tesseract_motion_planners/simple/profile). You can set the max cartesian and joint space allowed when interpolating.

closed 2023-06-11 by decdox