tesseract-robotics/tesseract_planning #322: Build issues with Ubuntu 22.04

srsidd, 2023-05-26:
Hi, I'm trying to build on Ubuntu 22.04 but running into compile errors. I'm using `0.16.3` on both `tesseract` and the `tesseract_planning` repo. I'm using colcon as the build tool.

```
In file included from /home/siddharth/Documents/tesseract_ws/install/tesseract_environment/include/tesseract_environment/environment.h:48,
                 from /home/siddharth/Documents/tesseract_ws/src/tesseract_planning/tesseract_motion_planners/descartes/include/tesseract_motion_planners/descartes/impl/descartes_motion_planner.hpp:39,
                 from /home/siddharth/Documents//tesseract_ws/src/tesseract_planning/tesseract_motion_planners/descartes/src/descartes_motion_planner.cpp:26:
/home/siddharth/Documents/tesseract_ws/install/tesseract_common/include/tesseract_common/manipulator_info.h:71:8: error: ‘variant’ in namespace ‘std’ does not name a template type
   71 |   std::variant<std::string, Eigen::Isometry3d> tcp_offset{ Eigen::Isometry3d::Identity() };
      |        ^~~~~~~
/home/siddharth/Documents/tesseract_ws/install/tesseract_common/include/tesseract_common/manipulator_info.h:71:3: note: ‘std::variant’ is only available from C++17 onwards
   71 |   std::variant<std::string, Eigen::Isometry3d> tcp_offset{ Eigen::Isometry3d::Identity() };
      |   ^~~
In file included from /home/siddharth/Documents/tesseract_ws/src/tesseract_planning/tesseract_motion_planners/descartes/include/tesseract_motion_planners/descartes/impl/descartes_motion_planner.hpp:39,
                 from /home/siddharth/Documents/tesseract_ws/src/tesseract_planning/tesseract_motion_planners/descartes/src/descartes_motion_planner.cpp:26:
/home/siddharth/Documents/tesseract_ws/install/tesseract_environment/include/tesseract_environment/environment.h:513:25: error: ‘shared_mutex’ is not a member of ‘std’
  513 |   std::shared_lock<std::shared_mutex> lockRead() const;
      |                         ^~~~~~~~~~~~
/home/siddharth/Documents/tesseract_ws/install/tesseract_environment/include/tesseract_environment/environment.h:513:25: note: ‘std::shared_mutex’ is only available from C++17 onwards
/home/siddharth/Documents/tesseract_ws/install/tesseract_environment/include/tesseract_environment/environment.h:513:25: error: ‘shared_mutex’ is not a member of ‘std’
/home/siddharth/Documents/tesseract_ws/install/tesseract_environment/include/tesseract_environment/environment.h:513:25: note: ‘std::shared_mutex’ is only available from C++17 onwards
/home/siddharth/Documents/tesseract_ws/install/tesseract_environment/include/tesseract_environment/environment.h:513:37: error: template argument 1 is invalid
  513 |   std::shared_lock<std::shared_mutex> lockRead() const;
```

Any leads on how to fix this?

Levi-Armstrong, 2023-05-27:
Can you try pulling the latest descartes_light and see if the issue goes away?

srsidd, 2023-05-29:
I found a hack / workaround from google that seemed to work. I added a snippet in `tesseract_motion_planners/CmakeLists.txt` to force the usage of C++17 - 
```
if(${CMAKE_VERSION} VERSION_LESS "3.8.0")
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17")
	unset(CMAKE_CXX_STANDARD) # Required not to override the flag above to c++14 or lower
else()
	if ("${CMAKE_CXX_STANDARD}" STREQUAL "" OR "${CMAKE_CXX_STANDARD}" LESS 17)
		set(CMAKE_CXX_STANDARD 17)
	endif()
endif()

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```
I was using 0.3.1 on `decartes_light`. I will try `master`.

srsidd, 2023-05-29:
On a semi-related note, I had installed taskflow on 20.04, using `sudo apt install taskflow` but 22.04 doesn't have a released binary for taskflow. How did you get around that?

marip8, 2023-05-30:
We have a binary build of `taskflow` and a few other packages on the [ROS-I PPA](https://launchpad.net/~ros-industrial/+archive/ubuntu/ppa)

lanmo0923, 2023-06-10:
@srsidd hello,I encountered the same problem on 22.04, how did you solve it?

acbuynak, 2023-06-11:
`taskflow` has not been released for Ubuntu 22 on the ROS-I PPA at https://launchpad.net/~ros-industrial/+archive/ubuntu/ppa

@Levi-Armstrong - Is there any reason to not add this to PPA?

`taskflow` can be built locally from source. Clone https://github.com/taskflow/taskflow.git into your workspace/src folder. I've been using `v3.4.0`. Build like normal using `colcon`.

If you use [vcstool](https://github.com/dirk-thomas/vcstool) or similar, I suggest adding a line like below to a dependencies.repos file.
```
- git
    local-name: taskflow
    uri: https://github.com/taskflow/taskflow.git
    version: v3.4.0
```

Levi-Armstrong, 2023-06-12:
I can work on releasing this on 22.04

Levi-Armstrong, 2023-06-12:
I just pushed the debian build to launchpad for Jammy so I will post in this thread when it has been published.

Levi-Armstrong, 2023-06-12:
@srsidd @lanmo0923 @acbuynak  Taskflow has been published on Jammy so you should be able to use the PPA to install taskflow

closed 2023-08-15 by Levi-Armstrong