tesseract-robotics/tesseract_planning #330: Remove `results` field from `TaskComposerNodeInfo`

marip8, 2023-06-12:
The `TaskComposerNodeInfo` struct has a field called `results`, presumably to store the results of the task. I think this member has been conceptually replaced by the `TaskComposerDataStorage` object with the specifiable `output_key`s in the `TaskComposerNodeInfo` struct. We should remove the `TaskComposerNodeInfo::result` member in favor of storing the results of the task only in the `TaskComposerDataStorage` object

https://github.com/tesseract-robotics/tesseract_planning/blob/dd5f9ae965061dce4b1a078e3bad575dcd838331/tesseract_task_composer/core/include/tesseract_task_composer/core/task_composer_node_info.h#L85-L86

Levi-Armstrong, 2023-06-12:
This is similar but different. The data_storage does not have to store the output of each task under unique keys. They are allowed to overwrite the contents under a key to limit data storage. The NodeInfo result is to be used for debug the results are explicitly saved here when requested. After looking at this I noticed this currently not being used because I never implemented the save_io functional back after the refactor. This could be removed for the base class and then in the planning we could add the input and output variable when we add back the save_io functionality.

marip8, 2023-06-12:
I think the existing infrastructure in the `TaskInfoDataStorage` class is totally adequate for debugging. If you want to see the results of each task, just don't overwrite output keys and all the results get saved. We may have to change some code to actually do this, but it's conceptually possible. I don't think there is any need to keep `TaskComposerNodeInfo::result` or to reintroduce functions for saving task-specific outputs when we could just leverage `TaskInfoDataStorage` directly

Levi-Armstrong, 2023-06-14:
I am fine with removing for now.

marip8, 2023-06-14:
It would also be really helpful to make a GUI widget that could display the inputs/outputs of a `TaskInfoDataStorage` object so you could see the state of the planned trajectories at each point in the pipeline

closed 2023-11-06 by Levi-Armstrong