tesseract-robotics/tesseract_planning #358: RasterFtPipeline showing odd behavior

peik-etzell, 2023-08-11:
I have been trying to do some planning with the RasterFtPipeline in tesseract_task_composer, and found some suspect behavior:

1. Transitions 1 and 2 are always completely broken, as if the _transition_ part of the trajectory was replaced by the $freespace_{start\rightarrow 1}$ / $freespace_{n \rightarrow end}$ motion
1. $Transition_{i \rightarrow i+1}$ seems to be queued for computation before $raster_i$ and $raster_{i+1}$ are finished computing, even though it is obviously dependent on the output of their computations (it seems the transition does wait for raster _i_, but I am not certain of anything here)
    - The freespace-task seems to plan to the output of the MinLengthTask as I understand it, which means there is a huge joint-interpolated jump in some cases
    - My testing has been purposefully _underconstrained_ (3-5 dof pose target, 9 dof robot), which makes this problem apparent, I suspect it might go unnoticed if the start- and end-states are fully constrained
1. _Sometimes_ the $freespace_{n \rightarrow end}$ motion is also bugged, jumping to one of the raster<>raster transitions

See the attached video, some notes:
- Transitions 1 and 2 broken
- Transition 3 plans without final target position -> joint-interpolated jump
- Transition 4 is perfect
- Transitions 5 and 6 do the same as T3, even though the jump is less noticable
- Transition 7 is again perfect

https://github.com/tesseract-robotics/tesseract_planning/assets/51529080/882dad5c-6211-4ac7-8d6f-e761beb375b0

I have made an as-minimal-as-possible [repo](https://github.com/peik-etzell/tesseract_raster_demo) that produces the above video for reproducibility.

Levi-Armstrong, 2023-08-11:
Than you for reporting and I will take look. Have you tried outputting the result dot graph?

Levi-Armstrong, 2023-08-11:
@marip8 and @marrts Have you been using the raster planners?

peik-etzell, 2023-08-11:
Ah yes I forgot to mention some observations from the dotgraph. I'll have to take a look at the dotgraph from this specifically later today, but testing our own robot I found that one transition subgraph was (always?) left white (uncomputed?), and this uncomputed one seemed to always be dependent on two [1] outputs from the rasters (dotted arrows), while all other transitions got one [0] and one [1], or two [0]. What I read in the source it is because the rasters are conditional, but this model seems incompatible with the 1 to many usage here?

Levi-Armstrong, 2023-08-13:
Thank you for posting the dot graph which made it easy to identify the issue. I will have a PR soon to fix the issue.

Levi-Armstrong, 2023-08-14:
@peik-etzell Do you mind testing my PR to make sure it solves the issue? I am going to update the unit tests so this is caught in the future.

peik-etzell, 2023-08-14:
Thank you @Levi-Armstrong ! This fixed the jumpy transitions for # > 2, from_start and transition # 2 are still broken though:

https://github.com/tesseract-robotics/tesseract_planning/assets/51529080/71368a74-0730-45aa-899a-fceb71d6f5f3

[image]
I can't see anything weird in the dotgraph

peik-etzell, 2023-08-14:
Nice, thank you very much! This solved it 👍

Levi-Armstrong, 2023-08-15:
Addressed by #360

closed 2023-08-15 by Levi-Armstrong