tesseract-robotics/tesseract_planning #361: Turning on OMPL path simplification breaks Trajopt

peik-etzell, 2023-08-17:
If you turn on `ompl_profile->simplify = true;`, the OMPL output trajectory is somehow broken, after which TrajOpt can't continue:

`[...] ERROR in validate_data: Lower bound at index 183 is greater than upper bound: -2.0942e+00 > -2.9667e+00`
_(close to the end of the output log)_

Should I post the same in the OMPL repo? (i.e. do you think this is a bug in OMPL or tesseract?)

Ran this: [https://github.com/peik-etzell/tesseract_raster_demo/tree/bug/OMPL-simplify](https://github.com/peik-etzell/tesseract_raster_demo/tree/bug/OMPL-simplify)

<details><summary>Debug output</summary>
<p>

[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2023-08-17-07-28-05-765000-ormond-287037
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [tesseract_raster_demo_node-1]: process started with pid [287039]
[tesseract_raster_demo_node-1] [INFO] [1692257286.079609774] [tesseract_raster_demo_node.ROSEnvironmentMonitor_internal.tesseract_ros_examples_monitor]: Updating internal planning scene state at most every 0.100000 seconds
[tesseract_raster_demo_node-1] [INFO] [1692257286.085043314] [tesseract_raster_demo_node.ROSEnvironmentMonitor_internal.tesseract_ros_examples_monitor]: Publishing maintained environment on '/tesseract_ros_examples/tesseract_published_environment'
[tesseract_raster_demo_node-1] [INFO] [1692257286.085146817] [tesseract_raster_demo_node.ROSEnvironmentMonitor_internal.tesseract_ros_examples_monitor]: Started environment state publishing thread ...
[tesseract_raster_demo_node-1] You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
[tesseract_raster_demo_node-1] Program: Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program: {
[tesseract_raster_demo_node-1] Program:   Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program:   {
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 1, State WP: Pos=0 0 0 0 0 0 0
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 1, Cart WP: xyz=0.4, -0.3, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:   }
[tesseract_raster_demo_node-1] Program:   Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program:   {
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.427952, -0.258568, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.451627, -0.214552, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.470791, -0.168393, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.48525, -0.120551, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.494861, -0.0715046, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.499527, -0.0217437, 0.6
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:   }
[tesseract_raster_demo_node-1] Program:   Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program:   {
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 1, Cart WP: xyz=0.499527, -0.0217437, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:   }
[tesseract_raster_demo_node-1] Program:   Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program:   {
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.499202, 0.0282344, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.49389, 0.0779304, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.483642, 0.126848, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.468562, 0.174498, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.448801, 0.220404, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 0, Cart WP: xyz=0.424555, 0.264108, 0.65
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:   }
[tesseract_raster_demo_node-1] Program:   Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program:   {
[tesseract_raster_demo_node-1] Program:     Move Instruction, Move Type: 1, Cart WP: xyz=0.424555, 0.264108, 0.7
[tesseract_raster_demo_node-1] , Description: Tesseract Move Instruction
[tesseract_raster_demo_node-1] Program:   }
[tesseract_raster_demo_node-1] Program:   Composite Instruction, Description: Tesseract Composite Instruction
[tesseract_raster_demo_node-1] Program:   {
[... 455 lines cut ...]
[tesseract_raster_demo_node-1] 0	(123)	|15129|
[tesseract_raster_demo_node-1] 1	(123)	|15129|
[tesseract_raster_demo_node-1] 2	(123)	|15129|
[tesseract_raster_demo_node-1] 3	(123)	|15129|
[tesseract_raster_demo_node-1] 4	(123)	|15129|
[tesseract_raster_demo_node-1] 5	(123)	|15129|
[tesseract_raster_demo_node-1] 6	(123)	|15129|
[tesseract_raster_demo_node-1] 7	(123)	|15498|
[tesseract_raster_demo_node-1] 8	(126)	|15876|
[tesseract_raster_demo_node-1] 9	(126)	|15372|
[tesseract_raster_demo_node-1] 10	(122)	|15006|
[tesseract_raster_demo_node-1] 11	(123)	|15129|
[tesseract_raster_demo_node-1] 12	(123)	|0|
[tesseract_raster_demo_node-1] 
[tesseract_raster_demo_node-1] 
[tesseract_raster_demo_node-1] Info:    No failed vertices
[tesseract_raster_demo_node-1] Info:    No failed edges
[tesseract_raster_demo_node-1] Debug:   Descartes took 0.0002 seconds to search graph for solution with cost 18.6676.
[tesseract_raster_demo_node-1] Debug:   Motion Planner process succeeded
[tesseract_raster_demo_node-1] Error:   Raster subgraph failed
[tesseract_raster_demo_node-1]          at line 477 in /workspaces/vscode_ros2_workspace/raster_demo_ws/src/tesseract_planning/tesseract_task_composer/planning/src/nodes/raster_motion_task.cpp
[tesseract_raster_demo_node-1] [INFO] [1692257287.585606236] [tesseract_raster_demo_node.ROSEnvironmentMonitor_internal.tesseract_ros_examples_monitor]: Published the Tesseract Environment State for: 'kuka_lbr_iiwa_14_r820'

</p>
</details>

Levi-Armstrong, 2023-08-17:
That is expected behavior. The simplify option was added for the use case where you are using OMPL only. In your case if you chaining OMPL to Trajopt this should be set to false. Internally because Tesseract want to try and maintain the input program structure at the minimum size. It will do simplification and/or interpolation to try and maintain the original size of the program. The program should never be shorter but can be long if necessary.

peik-etzell, 2023-08-17:
Alright, thank you! :+1:

closed 2023-08-17 by peik-etzell