tesseract-robotics/tesseract_planning #372: Using Convex Decomposition for Collision Geometry

emfebert, 2023-09-05:
I'm trying to import a custom workcell environment, is there a way to use VHACD for convex composition? I saw it's used somewhere in the code.

Here's the orginial workcell geometry:
[image]

And this is the result:
[image]

_Originally posted by @emfebert in https://github.com/tesseract-robotics/tesseract_planning/issues/369#issuecomment-1705746903_

rjoomen, 2023-09-05:
Here is some example code:

```
auto ConvexDecomposition(const tesseract_geometry::Mesh& mesh, int max_convex_hulls)
    -> std::vector<tesseract_geometry::ConvexMesh::Ptr>
{
  tesseract_collision::VHACDParameters params;
  params.find_best_plane = true;
  params.max_convex_hulls = max_convex_hulls;
  params.resolution = 1e6;
  tesseract_collision::ConvexDecompositionVHACD convex_decomp(params);
  std::vector<tesseract_geometry::ConvexMesh::Ptr> convex_meshes =
      convex_decomp.compute(*mesh.getVertices(), *mesh.getFaces());
  return convex_meshes;
}
```

Levi-Armstrong, 2023-09-05:
It should generate a command line tool that you can run to avoid having to write any code. I will take a look and make sure this is true.

Levi-Armstrong, 2023-09-05:
The executable is called `create_convex_hull` which you can run the with `-h` to get information on how to run it.

Levi-Armstrong, 2023-09-05:
Actually I think that only creates a convex hull and I don't see one for the convex decomposition.

emfebert, 2023-09-07:
Thanks for the pointers! I ended up using the original V-HACD repo https://github.com/kmammou/v-hacd since it was simple and quick to do.

closed 2023-09-07 by emfebert