tesseract-robotics/tesseract_planning #386: Error when adding Ruckig Smoothing to TrajoptTask

emfebert, 2023-09-28:
Hi,

I'd like perform trajectory interpolation on the trajectory generated by TrajOpt using Ruckig, but I'm running into an error.

Here's the segment of the yaml file configuring the TrajOptTask:

```
TrajOptTask:
        class: PipelineTaskFactory
        config:
          conditional: true
          inputs: [input_data]
          outputs: [output_data]
          nodes:
            DoneTask:
              class: DoneTaskFactory
              config:
                conditional: false
            ErrorTask:
              class: ErrorTaskFactory
              config:
                conditional: false
            MinLengthTask:
              class: MinLengthTaskFactory
              config:
                conditional: true
                inputs: [input_data]
                outputs: [output_data]
            TrajOptMotionPlannerTask:
              class: TrajOptMotionPlannerTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
                format_result_as_input: false
            DiscreteContactCheckTask:
              class: DiscreteContactCheckTaskFactory
              config:
                conditional: true
                inputs: [output_data]
            IterativeSplineParameterizationTask:
              class: IterativeSplineParameterizationTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
            RuckigTrajectorySmoothingTask:
              class: RuckigTrajectorySmoothingTaskFactory
              config:
                conditional: true
                inputs: [input_data]
                outputs: [output_data]
          edges:
            - source: MinLengthTask
              destinations: [ErrorTask, TrajOptMotionPlannerTask]
            - source: TrajOptMotionPlannerTask
              destinations: [ErrorTask, DiscreteContactCheckTask]
            - source: DiscreteContactCheckTask
              destinations: [ErrorTask, IterativeSplineParameterizationTask]
            - source: IterativeSplineParameterizationTask
              destinations: [ErrorTask, RuckigTrajectorySmoothingTask]
            - source: RuckigTrajectorySmoothingTask
              destinations: [ErrorTask, DoneTask]
          terminals: [ErrorTask, DoneTask]

```

And this is the error I'm getting:
```
Debug:   Profile 'FREESPACE' was not found in namespace 'RuckigTrajectorySmoothingTask' for type 'N18tesseract_planning41RuckigTrajectorySmoothingCompositeProfileE'. Using default if available. Available profiles:
Debug:   Profile 'FREESPACE' was not found in namespace 'RuckigTrajectorySmoothingTask' for type 'N18tesseract_planning36RuckigTrajectorySmoothingMoveProfileE'. Using default if available. Available profiles:
Debug:   Profile 'FREESPACE' was not found in namespace 'RuckigTrajectorySmoothingTask' for type 'N18tesseract_planning36RuckigTrajectorySmoothingMoveProfileE'. Using default if available. Available profiles:
tesseract_ros_examples_xarm7_trajopt_node: /home/febert/tesseract_ws2/src/tesseract_planning/tesseract_time_parameterization/core/src/instructions_trajectory.cpp:94: virtual Eigen::VectorXd& tesseract_planning::InstructionsTrajectory::getVelocity(Eigen::Index): Assertion `trajectory_[static_cast<std::size_t>(i)].get().as<MoveInstructionPoly>().getWaypoint().isStateWaypoint()' failed.

```

Thanks a lot!

emfebert, 2023-09-28:
I was able to pin down that the error is occuring here:
https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_time_parameterization/ruckig/src/ruckig_trajectory_smoothing.cpp#L316

Any ideas why this is happening?

rjoomen, 2023-09-29:
It seems you need to set `inputs: [output_data]` for Rucking, just like the node before it.

emfebert, 2023-09-29:
Thanks it works now!

I had another question: I'd like to get a fine interpolation (either joint or eef space) so I can run the trajectory on the robot. The above pipeline only provides 10 waypoints.

I tried inserting an Upsample task but it seems abort after it and isn't able to proceed with IterativeSplineParametrizationTask:

```
TrajOptTask:
        class: PipelineTaskFactory
        config:
          conditional: true
          inputs: [input_data]
          outputs: [output_data]
          nodes:
            DoneTask:
              class: DoneTaskFactory
              config:
                conditional: false
            ErrorTask:
              class: ErrorTaskFactory
              config:
                conditional: false
            MinLengthTask:
              class: MinLengthTaskFactory
              config:
                conditional: true
                inputs: [input_data]
                outputs: [output_data]
            TrajOptMotionPlannerTask:
              class: TrajOptMotionPlannerTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
                format_result_as_input: false
            DiscreteContactCheckTask:
              class: DiscreteContactCheckTaskFactory
              config:
                conditional: true
                inputs: [output_data]
            UpsampleTrajectoryTask:
              class: UpsampleTrajectoryTaskFactory
              config:
                conditional: false
                inputs: [output_data]
                outputs: [output_data]
            IterativeSplineParameterizationTask:
              class: IterativeSplineParameterizationTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
            RuckigTrajectorySmoothingTask:
              class: RuckigTrajectorySmoothingTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
          edges:
            - source: MinLengthTask
              destinations: [ErrorTask, TrajOptMotionPlannerTask]
            - source: TrajOptMotionPlannerTask
              destinations: [ErrorTask, DiscreteContactCheckTask]
            - source: DiscreteContactCheckTask
              destinations: [ErrorTask, UpsampleTrajectoryTask]
            - source: UpsampleTrajectoryTask
              destinations: [ErrorTask, IterativeSplineParameterizationTask]
            - source: IterativeSplineParameterizationTask
              destinations: [ErrorTask, RuckigTrajectorySmoothingTask]
            - source: RuckigTrajectorySmoothingTask
              destinations: [ErrorTask, DoneTask]
          terminals: [ErrorTask, DoneTask]

```

Is the upsampling Task doing interpolation in the joint space?

Thanks!

Levi-Armstrong, 2023-09-29:
If you are using ros this is typically done using one of the opensource ROS drivers for the robot. You can using ROS Control which some of the drivers use and manage the communication with the robot yourself.

emfebert, 2023-09-30:
Thanks a lot!

closed 2023-09-30 by emfebert