tesseract-robotics/tesseract_planning #390: Consistent naming: plan profile or move profile?

rjoomen, 2023-10-02:
What is the correct naming for this type of profile: plan profile or move profile? The naming in the code is inconsistent, see e.g. [here](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/motion_planner_task.hpp#L149).

I'm willing to refactor this everywhere, once I know the correct term.

Levi-Armstrong, 2023-10-02:
I would say move. The initial implementation we had plan_instruction and move_instruction, but removed the plan instruction a while back.

Levi-Armstrong, 2023-10-06:
@marip8 and @marrts What do you think?

marip8, 2023-10-06:
I think it's still on our roadmap (mental and currently undocumented) to do a refactor of the profiles to provide a common interface that they all should inherit from. I've started this in the `feat/ProfileRefactor` branch; [here](https://github.com/tesseract-robotics/tesseract_planning/blob/feat/ProfileRefactor/tesseract_motion_planners/core/include/tesseract_motion_planners/core/profiles.h) (from #267) I've created the three types of interfaces we think generically cover the use-cases of each planner: `WaypointProfile`, `CompositeProfile`, and `PlannerProfile`. What we're discussing here aligns with what I'm calling the `WaypointProfile`. I think this type of profile more generally applies the configuration of a planner for a `MoveInstruction` (since it's API takes a `MoveInstruction` as input), so I would vote for renaming to `MoveProfile`.

In digging up this information, I noticed that there does not appear to be an issue or discussion about the profile refactor on this repo even though @Levi-Armstrong and I have discussed it quite a bit in conversations in the past. I'll try to collect all of our thoughts about this and put it into a discussion for documentation and discussion

rjoomen, 2023-10-06:
For reference: #260 and linked work: #255 and #367

closed 2025-04-23 by rjoomen