tesseract-robotics/tesseract_planning #396: Expected behavior issue with ErrorTask

dpiet, 2023-10-13:
In 0.20.x recent releases, looks like `AbortTask` was completely removed and somewhat replaced with `ErrorTask`, per commit `dacd8a4`. It adds an abort field that defaults to false. Does it make sense to default this to true to maintain expected behavior? It's not mentioned in release notes and we only came across the problem after system testing and reviewing updated README doc. Many API changes have broken things recently and I would suggest moving to a policy of more deprecation warnings in advance of complete removals.

Levi-Armstrong, 2023-10-13:
We currently use the minor version to indicate if breaking changes have occurred within Tesseract because a 1.0.0 release has not been achieved. This is to be expected if you are upgrading to a higher minor version.

Levi-Armstrong, 2023-10-13:
>Does it make sense to default this to true to maintain expected behavior?

No, because an error is an error does not indicate if the whole pipeline should be aborted. This may just cause your pipeline to go down a different branch instead aborting the pipeline. Every task can now trigger an abort if reached and adding `trigger_abort: true`.

dpiet, 2023-10-15:
Yep I totally understand the rationale for the task change and minor version policies. However there has been a very rapid cadence of releases the past few months and bug fixes are not getting back ported. And with the refactors there have been some serious bugs that forced us to upgrade. I'm just suggesting we add some more stability so folks can build their applications. I'll close this issue since you've addressed it and we can move to discussion.

closed 2023-10-15 by dpiet

Levi-Armstrong, 2023-10-15:
I try to do patch release with bug fixes but as you mentioned that things are moving fast. If you are on a particular release and a bug fix was not back ported and you do not want to upgrade yet, you can create a PR base on your release I would be happy to merge into a patch branch and create a new tagged version for that release.