tesseract-robotics/tesseract_planning #397: Different profile string in CompositeInstruction leads to totally diffenent planning results

thume4zzzz, 2023-10-20:
Hi, I'm using trajopt to plan a trajectory subject to Cartesian waypoint constraints. 
If I use the following codes to create the planning program, everything works fine. 
```
// Create Program
CompositeInstruction program("DEFAULT", CompositeInstructionOrder::ORDERED, mi);
```

However, if I follow the example in _basic_cartesian_example.cpp_ and use:
```
CompositeInstruction program(
      "cartesian_program", CompositeInstructionOrder::ORDERED, ManipulatorInfo("manipulator", "base_link", "tool0"));
```

the planner will fail because some of the waypoint constraints can not be satisfied.  
[image]

Could someone tell me what is the difference?

marip8, 2023-10-20:
In the code snippet from this example, we are configuring a TrajOpt profile that is specific to this application under the name "cartesian_program", and it has different values than the default values of the `TrajOptDefaultCompositeProfile` class. Therefore, when you change the composite instruction to use the `DEFAULT` profile, you're telling the planner to apply various costs and constraints that may not really be appropriate

https://github.com/tesseract-robotics/tesseract_planning/blob/1c6104eb782ebbc85e116cd7a32df3a74d6c25e2/tesseract_examples/src/basic_cartesian_example.cpp#L224-L231

If you could post the full TrajOpt log from the last iteration of the optimization, we may be able to identify what costs/constraints were causing the problem

thume4zzzz, 2023-10-25:
Thank you for your reply. I create a custom cartesian_program just like the example, now it works well. Here is another question. I wrote a test program that commands the manipulator to follow a circular path in Cartesian space. 
```
  int point_num = 11;
  Eigen::Vector3d center(x,y,z);
  for (std::size_t i = 0; i < point_num; ++i)
  {
    theta =  2*M_PI / (point_num - 1) * i ;

    Eigen::Vector3d end(x-r*sin(theta), y, z+r*cos(theta));
    Eigen::Vector3d norm_z = (end - center).normalized();
    Eigen::Vector3d x_axis(0, 1, 0); 
    Eigen::Vector3d y_axis = (norm_z.cross(x_axis)).normalized();
    Eigen::Isometry3d pose;
    pose.matrix().col(0).head<3>() = x_axis;
    pose.matrix().col(1).head<3>() = y_axis;
    pose.matrix().col(2).head<3>() = norm_z;
    pose.matrix().col(3).head<3>() = end;

    CartesianWaypointPoly wp{ CartesianWaypoint(pose) };
    MoveInstruction plan_instruction(wp, MoveInstructionType::FREESPACE, "freespace_profile");
    plan_instruction.setDescription("waypoint_" + std::to_string(i));
    program.appendMoveInstruction(plan_instruction);
  }
```
The planner can correctly solve the trajectory.  However, if I change the point_num to 10 or 20, the planner will fail to find a solution. What is the problem?

thume4zzzz, 2023-10-25:
Here is the TrajOpt log from the last iteration of the optimization. The cartesian poses of planned trajectory deviate greatly from the set values.

| ======================================================================================== |
|                                      ROS Industrial                                      |
|                                  TrajOpt Motion Planning                                 |
| ======================================================================================== |
|                            Iteration (Box Size: 0.001500000)                             |
| ---------------------------------------------------------------------------------------- |
|        Overall: 11   |      Convexify: 1    |    Trust Region: 1   |        Penalty: 4   |
| ======================================================================================== |
|      merit |   oldexact |  new_exact | new_approx |    dapprox |     dexact |      ratio |
| ---------------------------------------------------------------------------------------- | INDIVIDUAL COSTS
| ---------- |  1.076e-04 |  1.083e-04 |  1.083e-04 | -6.853e-07 | -6.853e-07 |  1.000e+00 | JointVelocity_0
| ---------- |  1.076e-04 |  1.083e-04 |  1.083e-04 | -7.192e-07 | -7.192e-07 |  1.000e+00 | JointVelocity_1
| ---------- |  1.083e-04 |  1.090e-04 |  1.090e-04 | -7.101e-07 | -7.101e-07 |  1.000e+00 | JointVelocity_2
| ---------- |  1.076e-04 |  1.083e-04 |  1.083e-04 | -7.194e-07 | -7.194e-07 |  1.000e+00 | JointVelocity_3
| ---------- |  1.083e-04 |  1.093e-04 |  1.093e-04 | -1.004e-06 | -1.004e-06 |  1.000e+00 | JointVelocity_4
| ---------- |  1.076e-04 |  1.083e-04 |  1.083e-04 | -7.209e-07 | -7.209e-07 |  1.000e+00 | JointVelocity_5
| ---------- |  1.083e-04 |  1.096e-04 |  1.096e-04 | -1.300e-06 | -1.300e-06 |  1.000e+00 | JointVelocity_6
| ---------- |  1.076e-04 |  1.083e-04 |  1.083e-04 | -7.188e-07 | -7.188e-07 |  1.000e+00 | JointVelocity_7
| ---------- |  3.543e-13 |  6.704e-19 |  6.704e-19 |  3.543e-13 |  3.543e-13 |   ------   | JointVelocity_8
| ---------- |  1.643e-13 |  1.636e-13 |  1.636e-13 |  7.770e-16 |  7.770e-16 |   ------   | JointVelocity_9
| ---------- |  7.004e-14 |  6.989e-14 |  6.989e-14 |  1.433e-16 |  1.433e-16 |   ------   | JointVelocity_10
| ---------- |  1.267e-10 |  8.356e-11 |  8.356e-11 |  4.311e-11 |  4.311e-11 |   ------   | JointVelocity_11
| ---------- |  6.521e-14 |  6.507e-14 |  6.507e-14 |  1.455e-16 |  1.455e-16 |   ------   | JointVelocity_12
| ---------- |  1.680e-10 |  3.241e-10 |  3.241e-10 | -1.561e-10 | -1.561e-10 |   ------   | JointVelocity_13
| ---------- |  2.120e-14 |  2.107e-14 |  2.107e-14 |  1.267e-16 |  1.267e-16 |   ------   | JointVelocity_14
| ---------- |  2.383e-10 |  7.655e-10 |  7.655e-10 | -5.272e-10 | -5.272e-10 |   ------   | JointVelocity_15
| ---------- |  3.448e-14 |  3.430e-14 |  3.430e-14 |  1.718e-16 |  1.718e-16 |   ------   | JointVelocity_16
| ---------- |  5.407e-15 |  3.273e-20 |  3.273e-20 |  5.407e-15 |  5.407e-15 |   ------   | JointVelocity_17
| ---------- |  3.030e-14 |  3.016e-14 |  3.016e-14 |  1.433e-16 |  1.433e-16 |   ------   | JointVelocity_18
| ---------- |  1.376e-14 |  1.364e-14 |  1.364e-14 |  1.252e-16 |  1.252e-16 |   ------   | JointVelocity_19
| ---------- |  3.666e-10 |  4.598e-11 |  4.598e-11 |  3.206e-10 |  3.206e-10 |   ------   | JointVelocity_20
| ---------- |  1.291e-14 |  1.279e-14 |  1.279e-14 |  1.202e-16 |  1.202e-16 |   ------   | JointVelocity_21
| ---------- |  3.884e-10 |  1.295e-10 |  1.295e-10 |  2.589e-10 |  2.589e-10 |   ------   | JointVelocity_22
| ---------- |  5.103e-15 |  5.037e-15 |  5.037e-15 |  6.538e-17 |  6.538e-17 |   ------   | JointVelocity_23
| ---------- |  4.062e-10 |  2.520e-10 |  2.520e-10 |  1.542e-10 |  1.542e-10 |   ------   | JointVelocity_24
| ---------- |  1.099e-14 |  1.089e-14 |  1.089e-14 |  9.977e-17 |  9.977e-17 |   ------   | JointVelocity_25
| ---------- |  4.959e-14 |  1.373e-19 |  1.373e-19 |  4.959e-14 |  4.959e-14 |   ------   | JointVelocity_26
| ---------- |  3.097e-16 |  3.144e-16 |  3.144e-16 | -4.685e-18 | -4.685e-18 |   ------   | JointVelocity_27
| ---------- |  7.022e-17 |  7.403e-17 |  7.403e-17 | -3.812e-18 | -3.812e-18 |   ------   | JointVelocity_28
| ---------- |  3.627e-10 |  5.281e-11 |  5.281e-11 |  3.099e-10 |  3.099e-10 |   ------   | JointVelocity_29
| ---------- |  8.618e-17 |  9.036e-17 |  9.036e-17 | -4.179e-18 | -4.179e-18 |   ------   | JointVelocity_30
| ---------- |  3.495e-10 |  8.156e-11 |  8.156e-11 |  2.680e-10 |  2.680e-10 |   ------   | JointVelocity_31
| ---------- |  6.700e-16 |  6.796e-16 |  6.796e-16 | -9.614e-18 | -9.614e-18 |   ------   | JointVelocity_32
| ---------- |  3.385e-10 |  1.176e-10 |  1.176e-10 |  2.208e-10 |  2.208e-10 |   ------   | JointVelocity_33
| ---------- |  2.301e-17 |  2.490e-17 |  2.490e-17 | -1.894e-18 | -1.894e-18 |   ------   | JointVelocity_34
| ---------- |  4.023e-14 |  3.565e-20 |  3.565e-20 |  4.023e-14 |  4.023e-14 |   ------   | JointVelocity_35
| ---------- |  4.388e-15 |  4.394e-15 |  4.394e-15 | -5.407e-18 | -5.407e-18 |   ------   | JointVelocity_36
| ---------- |  1.729e-14 |  1.730e-14 |  1.730e-14 | -1.467e-17 | -1.467e-17 |   ------   | JointVelocity_37
| ---------- |  1.553e-10 |  4.853e-11 |  4.853e-11 |  1.068e-10 |  1.068e-10 |   ------   | JointVelocity_38
| ---------- |  1.716e-14 |  1.718e-14 |  1.718e-14 | -1.441e-17 | -1.441e-17 |   ------   | JointVelocity_39
[... cut ...]
Info:    Not all constraints are satisfied. Increasing constraint penalties for 122
Info:    Not all constraints are satisfied. Increasing constraint penalties for 123
Info:    Not all constraints are satisfied. Increasing constraint penalties for 124
Info:    Not all constraints are satisfied. Increasing constraint penalties for 125
Info:    Not all constraints are satisfied. Increasing constraint penalties for 127
Info:    Not all constraints are satisfied. Increasing constraint penalties for 128

marip8, 2023-10-25:
> Here is the TrajOpt log from the last iteration of the optimization. The cartesian poses of planned trajectory deviate greatly from the set values.

Two questions:
- Is this log from your custom application you described above or from `basic_cartesian_example.cpp`?
- Are you using the TrajOpt IFOPT planner or the "regular" TrajOpt planner? What version of the `trajopt` repo are you on?
  - I have only used the "regular" TrajOpt planner before, and I have not seen it create joint velocity costs for every set of decision variables as in your log; I also don't recognize the names of the costs and constraints

The optimization seems to have failed because all of the cartesian position costs (`CartPos_JointPosition` in the `new_exact` column) are very large but should be zero.

My guess as to why this fails is that you might not have given TrajOpt a good joint seed for those Cartesian positions, so it got stuck and wasn't able to find a way to meet those constraints. I would recommend adding a `SimplePlanner` step before going to TrajOpt. This planner will solve IK for each Cartesian waypoint and set that IK solution as the seed for TrajOpt. Then TrajOpt can optimize that IK solution (if your robot has more than 6 DoF) and will likely be more successful

thume4zzzz, 2023-10-26:
> > Here is the TrajOpt log from the last iteration of the optimization. The cartesian poses of planned trajectory deviate greatly from the set values.
> 
> A few questions:
> 
>     * Is this log from your custom application you described above or from `basic_cartesian_example.cpp`?
> 
>     * Are you using the TrajOpt IFOPT planner or the "regular" TrajOpt planner? What version of the `trajopt` repo are you on?
>       
>       * I have only used the "regular" TrajOpt planner before, and I have not seen it create joint velocity costs for every set of decision variables as in your log; I also don't recognize the names of the costs and constraints
> 
> 
> The optimization seems to have failed because all of the cartesian position costs (`CartPos_JointPosition` in the `new_exact` column) are very large but should be zero.
> 
> My guess as to why this fails is that you might not have given TrajOpt a good joint seed for those Cartesian positions, so it got stuck and wasn't able to find a way to meet those constraints. I would recommend adding a `SimplePlanner` step before going to TrajOpt. This planner will solve IK for each Cartesian waypoint and set that IK solution as the seed for TrajOpt. Then TrajOpt can optimize that IK solution (if your robot has more than 6 DoF) and will likely be more successful

- This log is from my custom application based on basic_cartesian_example.cpp (use the same cartesian_program).
- I'm using TrajOpt IFOPT and the latest version (0.6.1). 

Could you please give an example how to add a SimplePlanner?

thume4zzzz, 2023-10-26:
> > > Here is the TrajOpt log from the last iteration of the optimization. The cartesian poses of planned trajectory deviate greatly from the set values.
> > 
> > 
> > A few questions:
> > ```
> > * Is this log from your custom application you described above or from `basic_cartesian_example.cpp`?
> > 
> > * Are you using the TrajOpt IFOPT planner or the "regular" TrajOpt planner? What version of the `trajopt` repo are you on?
> >   
> >   * I have only used the "regular" TrajOpt planner before, and I have not seen it create joint velocity costs for every set of decision variables as in your log; I also don't recognize the names of the costs and constraints
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > The optimization seems to have failed because all of the cartesian position costs (`CartPos_JointPosition` in the `new_exact` column) are very large but should be zero.
> > My guess as to why this fails is that you might not have given TrajOpt a good joint seed for those Cartesian positions, so it got stuck and wasn't able to find a way to meet those constraints. I would recommend adding a `SimplePlanner` step before going to TrajOpt. This planner will solve IK for each Cartesian waypoint and set that IK solution as the seed for TrajOpt. Then TrajOpt can optimize that IK solution (if your robot has more than 6 DoF) and will likely be more successful
> 
>     * This log is from my custom application based on basic_cartesian_example.cpp (use the same cartesian_program).
> 
>     * I'm using TrajOpt IFOPT and the latest version trajopt repo (0.6.1).
> 
> 
> Could you please give an example how to add a SimplePlanner?

Update:
I follow the example in chain_example.cpp which utilizes descartes to get the seed for trajopt. But descartes fails to get monitor environment information. What should I do ?
[descartes_log.txt](https://github.com/tesseract-robotics/tesseract_planning/files/13172646/descartes_log.txt)

marip8, 2023-10-26:
If your robot is redundant and you are not discretely sampling the extra DoF, the `descartes` planner is not going to provide any additional benefit over the simple planner (i.e., interpolation and IK solving). If you notice in the `descartes` log, no waypoint (rung) in your plan has more than one IK solution (node), meaning that if all your waypoints were feasible there would only be one path through the graph and it would be equivalent to sequentially solving IK.

It appears the [example is using a pipeline that only includes TrajOpt IFOPT](). To use the simple planner in conjunction with TrajOpt IFOPT, you'll have to make a custom pipeline that is similar to [this one](https://github.com/tesseract-robotics/tesseract_planning/blob/d2b11debc8b6ab414c078333430db4004771445b/tesseract_task_composer/config/task_composer_plugins.yaml#L417). It would probably look like this:

```diff
 TrajOptIfoptTask:
        class: PipelineTaskFactory
        config:
          conditional: true
          inputs: [input_data]
          outputs: [output_data]
          nodes:
            DoneTask:
              class: DoneTaskFactory
              config:
                conditional: false
            ErrorTask:
              class: ErrorTaskFactory
              config:
                conditional: false
            MinLengthTask:
              class: MinLengthTaskFactory
              config:
                conditional: true
                inputs: [input_data]
                outputs: [output_data]
+            SimpleMotionPlannerTask:
+              class: SimpleMotionPlannerTaskFactory
+               config:
+                conditional: true
+                inputs: [input_data]
+                outputs: [output_data]
+                format_result_as_input: true
            TrajOptIfoptMotionPlannerTask:
              class: TrajOptIfoptMotionPlannerTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
                format_result_as_input: false
            DiscreteContactCheckTask:
              class: DiscreteContactCheckTaskFactory
              config:
                conditional: true
                inputs: [output_data]
            IterativeSplineParameterizationTask:
              class: IterativeSplineParameterizationTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
          edges:
            - source: MinLengthTask
-             destinations: [ErrorTask, TrajOptIfoptMotionPlannerTask]
+             destinations: [ErrorTask, SimpleMotionPlannerTask]
+           - source: SimpleMotionPlannerTask
+             destinations: [ErrorTask, TrajOptIfoptMotionPlannerTask]
            - source: TrajOptIfoptMotionPlannerTask
              destinations: [ErrorTask, DiscreteContactCheckTask]
            - source: DiscreteContactCheckTask
              destinations: [ErrorTask, IterativeSplineParameterizationTask]
            - source: IterativeSplineParameterizationTask
              destinations: [ErrorTask, DoneTask]
          terminals: [ErrorTask, DoneTask]
```

The simple planner has a few [profiles](https://github.com/tesseract-robotics/tesseract_planning/tree/master/tesseract_motion_planners/simple/include/tesseract_motion_planners/simple/profile) that determine how it performs interpolation. Either the LVS (default) or fixed size plan profiles should work here

thume4zzzz, 2023-10-27:
> If your robot is redundant and you are not discretely sampling the extra DoF, the `descartes` planner is not going to provide any additional benefit over the simple planner (i.e., interpolation and IK solving). If you notice in the `descartes` log, no waypoint (rung) in your plan has more than one IK solution (node), meaning that if all your waypoints were feasible there would only be one path through the graph and it would be equivalent to sequentially solving IK.
> 
> It appears the example is using a pipeline that only includes TrajOpt IFOPT. To use the simple planner in conjunction with TrajOpt IFOPT, you'll have to make a custom pipeline that is similar to [this one](https://github.com/tesseract-robotics/tesseract_planning/blob/d2b11debc8b6ab414c078333430db4004771445b/tesseract_task_composer/config/task_composer_plugins.yaml#L417). It would probably look like this:
> 
> ```diff
>  TrajOptIfoptTask:
>         class: PipelineTaskFactory
>         config:
>           conditional: true
>           inputs: [input_data]
>           outputs: [output_data]
>           nodes:
>             DoneTask:
>               class: DoneTaskFactory
>               config:
>                 conditional: false
>             ErrorTask:
>               class: ErrorTaskFactory
>               config:
>                 conditional: false
>             MinLengthTask:
>               class: MinLengthTaskFactory
>               config:
>                 conditional: true
>                 inputs: [input_data]
>                 outputs: [output_data]
> +            SimpleMotionPlannerTask:
> +              class: SimpleMotionPlannerTaskFactory
> +               config:
> +                conditional: true
> +                inputs: [input_data]
> +                outputs: [output_data]
> +                format_result_as_input: true
>             TrajOptIfoptMotionPlannerTask:
>               class: TrajOptIfoptMotionPlannerTaskFactory
>               config:
>                 conditional: true
>                 inputs: [output_data]
>                 outputs: [output_data]
>                 format_result_as_input: false
>             DiscreteContactCheckTask:
>               class: DiscreteContactCheckTaskFactory
>               config:
>                 conditional: true
>                 inputs: [output_data]
>             IterativeSplineParameterizationTask:
>               class: IterativeSplineParameterizationTaskFactory
>               config:
>                 conditional: true
>                 inputs: [output_data]
>                 outputs: [output_data]
>           edges:
> +           - source: SimpleMotionPlannerTask
> +             destinations: [ErrorTask, MinLengthTask]
>             - source: MinLengthTask
>               destinations: [ErrorTask, TrajOptIfoptMotionPlannerTask]
>             - source: TrajOptIfoptMotionPlannerTask
>               destinations: [ErrorTask, DiscreteContactCheckTask]
>             - source: DiscreteContactCheckTask
>               destinations: [ErrorTask, IterativeSplineParameterizationTask]
>             - source: IterativeSplineParameterizationTask
>               destinations: [ErrorTask, DoneTask]
>           terminals: [ErrorTask, DoneTask]
> ```
> 
> The simple planner has a few [profiles](https://github.com/tesseract-robotics/tesseract_planning/tree/master/tesseract_motion_planners/simple/include/tesseract_motion_planners/simple/profile) that determine how it performs interpolation. Either the LVS (default) or fixed size plan profiles should work here

Thank you for replying. 
I have modified the yaml file according to your advice. But how to add a simple planner in the application? I use the following code : 
```
  auto profiles = std::make_shared<ProfileDictionary>();

  // Simple planner profiles 
  auto simple_planner_profile = std::make_shared<SimplePlannerLVSPlanProfile>();
  profiles->addProfile<SimplePlannerPlanProfile>(SIMPLE_DEFAULT_NAMESPACE, "DEFAULT", simple_planner_profile);

  // Trajopt profiles 
  auto composite_profile = std::make_shared<TrajOptIfoptDefaultCompositeProfile>();
  composite_profile->collision_cost_config->type = tesseract_collision::CollisionEvaluatorType::LVS_DISCRETE;
  composite_profile->collision_constraint_config->type = tesseract_collision::CollisionEvaluatorType::LVS_DISCRETE;
  composite_profile->smooth_velocities = true;
  composite_profile->smooth_accelerations = false;
  composite_profile->smooth_jerks = false;
  composite_profile->velocity_coeff = Eigen::VectorXd::Ones(1);
  profiles->addProfile<TrajOptIfoptCompositeProfile>(
      TRAJOPT_IFOPT_DEFAULT_NAMESPACE, "cartesian_program", composite_profile);

  auto plan_profile = std::make_shared<TrajOptIfoptDefaultPlanProfile>();
  plan_profile->cartesian_coeff = Eigen::VectorXd::Ones(6);
  plan_profile->joint_coeff = Eigen::VectorXd::Ones(numJoints);
  profiles->addProfile<TrajOptIfoptPlanProfile>(TRAJOPT_IFOPT_DEFAULT_NAMESPACE, "RASTER", plan_profile);
  profiles->addProfile<TrajOptIfoptPlanProfile>(TRAJOPT_IFOPT_DEFAULT_NAMESPACE, "freespace_profile", plan_profile);
  
  // Create task
  const std::string task_name = "TrajOptIfoptTask";
  TaskComposerNode::UPtr task = factory.createTaskComposerNode(task_name);
  const std::string output_key = task->getOutputKeys().front();

  // Create Task Composer Problem
  auto problem = std::make_unique<PlanningTaskComposerProblem>(env_, profiles);
  problem->input_instruction = program;
```
but get 
Error:   Input instructions to MotionPlannerTask: SimpleMotionPlannerTask must be a composite instruction

marip8, 2023-10-30:
> But how to add a simple planner in the application? 

The simple planner is already a part of the `TrajOptIfoptTask` "pipeline", so it get's added as a part of `TaskComposerNode::UPtr task = factory.createTaskComposerNode("TrajOptIfoptTask");`.

> Error: Input instructions to MotionPlannerTask: SimpleMotionPlannerTask must be a composite instruction

This is exactly the problem described in #399. It appears that your problem is set up incorrectly. Per the discussion on that issue, the better way to solve this issue is to add a `ProcessInputTask` which looks at `problem->input_instruction` and makes sure that the information gets set in the correct place for the first task in the underlying `DataStorage` object:

```diff
 TrajOptIfoptTask:
        class: PipelineTaskFactory
        config:
          conditional: true
          inputs: [input_data]
          outputs: [output_data]
          nodes:
            DoneTask:
              class: DoneTaskFactory
              config:
                conditional: false
            ErrorTask:
              class: ErrorTaskFactory
              config:
                conditional: false
+            ProcessInputTask:
+              class: ProcessPlanningInputTaskFactory
+              config:
+                conditional: false
+                outputs: [input_data]
            MinLengthTask:
              class: MinLengthTaskFactory
              config:
                conditional: true
                inputs: [input_data]
                outputs: [output_data]
+            SimpleMotionPlannerTask:
+              class: SimpleMotionPlannerTaskFactory
+               config:
+                conditional: true
+                inputs: [input_data]
+                outputs: [output_data]
+                format_result_as_input: true
            TrajOptIfoptMotionPlannerTask:
              class: TrajOptIfoptMotionPlannerTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
                format_result_as_input: false
            DiscreteContactCheckTask:
              class: DiscreteContactCheckTaskFactory
              config:
                conditional: true
                inputs: [output_data]
            IterativeSplineParameterizationTask:
              class: IterativeSplineParameterizationTaskFactory
              config:
                conditional: true
                inputs: [output_data]
                outputs: [output_data]
          edges:
+           - source: ProcessInputTask
+           - destinations: [SimpleMotionPlannerTask]
+           - source: SimpleMotionPlannerTask
+             destinations: [ErrorTask, MinLengthTask]
            - source: MinLengthTask
              destinations: [ErrorTask, TrajOptIfoptMotionPlannerTask]
            - source: TrajOptIfoptMotionPlannerTask
              destinations: [ErrorTask, DiscreteContactCheckTask]
            - source: DiscreteContactCheckTask
              destinations: [ErrorTask, IterativeSplineParameterizationTask]
            - source: IterativeSplineParameterizationTask
              destinations: [ErrorTask, DoneTask]
          terminals: [ErrorTask, DoneTask]
```

thume4zzzz, 2023-11-01:
It seems like that KDL fails to generate joint solution for a reachable cartesian pose. Is it possible to switch to a different ik solver?

marip8, 2023-11-01:
> It seems like that KDL fails to generate joint solution for a reachable cartesian pose. Is it possible to switch to a different ik solver?

Depends on your robot. If it is a 6-DOF spherical wrist robot, you could use the OPW solver. If it's a UR, you can use the UR solvers. Otherwise, the KDL solvers are really the only option at the moment. You could update the seed state in your IK call to be closer to the final solution; that typically helps a lot with making the solver converge

thume4zzzz, 2023-11-02:
Thank you for your reply. Finally I figure it out by using Descartes + trajopt and update the seed state for KDL in each IK call. The following is the modified sample function in _descartes_robot_sample.hpp_:
```
  // Generate all possible Cartesian poses
  tesseract_common::VectorIsometry3d target_poses = target_pose_sampler_(target_pose_);

  bool found_ik_sol = false;
  std::stringstream error_string_stream;

  // Generate the IK solutions for those poses
  std::vector<descartes_light::StateSample<FloatType>> samples; 
  const auto limit = manip_->getLimits().joint_limits; 
  const int max_gen_seed = 50; 
  int seed_count = 0; 
  Eigen::VectorXd ik_seed(manip_->numJoints());

  // for (const auto& pose : target_poses)
  for (std::size_t i = 0; i < target_poses.size(); i++)
  {
    const auto& pose = target_poses[i];

    // Get the transformation to the kinematic tip link
    Eigen::Isometry3d target_pose = pose * tcp_offset_.inverse();

    // generate multiple ik seed for ik solutions 
    tesseract_kinematics::IKSolutions ik_solutions; 

    if (ik_seed_global.isZero())
    {
      while ( seed_count < max_gen_seed && ik_solutions.empty()){
        std::random_device rd;
        std::mt19937 gen(rd());   // Mersenne Twister 19937 random seed generator
        for (std::size_t j = 0; j < manip_->numJoints(); ++j){
          std::uniform_real_distribution<double> distribution_double(limit(j, 0), limit(j, 1));
          ik_seed(j) = distribution_double(gen);
        }
        
        tesseract_kinematics::KinGroupIKInput ik_input(target_pose, target_working_frame_, tcp_frame_);
        tesseract_kinematics::IKSolutions ik_solution = manip_->calcInvKin({ ik_input }, ik_seed);
        ik_solutions.insert(ik_solutions.end(), ik_solution.begin(), ik_solution.end());
      }
    }
    else
    {
      tesseract_kinematics::KinGroupIKInput ik_input(target_pose, target_working_frame_, tcp_frame_);
      tesseract_kinematics::IKSolutions ik_solution = manip_->calcInvKin({ ik_input }, ik_seed_global);
      ik_solutions.insert(ik_solutions.end(), ik_solution.begin(), ik_solution.end());
    }

    if (ik_solutions.empty())
      continue;
    else 
      ik_seed_global = ik_solutions[0];
```

thume4zzzz, 2023-11-03:
Thank you for you reply. Finally I figure it out by changing the seed for KDL in each IK call. I also turn off collision check in Descartes. Now Descartes + trajopt work well when 
`trajopt_composite_profile->collision_cost_config.enabled = false;`
[image]
But if I turn the collision cost on, trajopt will get stuck in the first Iteration.
[image]