tesseract-robotics/tesseract_planning #400: Patch 0.20.1 build

marip8, 2023-10-24:
Building 0.20.1 of this repo with 0.20.1 of `tesseract` results in a build failure because the header clean up from https://github.com/tesseract-robotics/tesseract/pull/946 removed some headers that were implicitly included in `tesseract_planning`. This issue was fixed in #394, but only after breaking changes were introduced in this repo to support a later version of `trajopt`.

What is the right way of patching the 0.20.1 build to include this fix?

Levi-Armstrong, 2023-10-24:
I would create a branch from the 0.20.1 tag and apply the patch then create a 0.20.2 tag from this branch then the branch can be deleted.

Levi-Armstrong, 2023-10-24:
I don't think you will be able to use the catkin_prepare_release but you should be able to use the catkin_generate_changelog to update the change log and then manually push to the patch branch and then create tag.

marip8, 2023-10-24:
> I would create a branch from the 0.20.1 tag and apply the patch then create a 0.20.2 tag from this branch then the branch can be deleted.

I thought of this last night as well; I'll give it a shot

marip8, 2023-10-24:
Closing; I think this should work fine

closed 2023-10-24 by marip8