tesseract-robotics/tesseract_planning #412: Erratic behavior when visualizing motion plan

emfebert, 2023-11-08:
Hi,

We're running trajopt planning through a series of waypoints, but when visualizing the trajectory we can see the robot making sudden 360 degree rotations on the base joint between several of the waypoints.

https://github.com/tesseract-robotics/tesseract_planning/assets/128838660/d7446a72-1bb0-4976-82b3-aaf214c7ad07

This looks like an interpolation error, perhaps it's clipping angles in erroneous ways. Any ideas what could be the root cause?

Thanks a lot!

Best,
Frederik

Levi-Armstrong, 2023-11-09:
I agree with you that it seems to be a bad seed trajectory. I recommend also saving out the trajectory returned from the simple planner within you pipeline to confirm.

Levi-Armstrong, 2025-01-04:
Reopen if still an issue.

closed 2025-01-04 by Levi-Armstrong